sm0svx/svxlink #347: Signal transmission, "Roger Beep" in "Listen Only" mode...

UR3QJW, 2018-01-06:
After connecting to the conference, in the network of EchoLink and activation, "Listen Only" mode ...
The SvxLink, continues to transmit to the EchoLink network, the signal of "Roger Beep".

Axel-DG1BBQ, 2018-01-07:
You can disable the roger beep that is sent out via EchoLink like this:
In EchoLink.tcl in the procedure proc squelch_open comment out too lines.

BEFORE:
=====
#
# Executed when the squelch state changes
#
proc squelch_open {is_open} {
  if {!$is_open} {
    #playSilence 200
    playTone 1000 100 100
  }
}
AFTER:
====
#
# Executed when the squelch state changes
#
proc squelch_open {is_open} {
  if {!$is_open} {
    #playSilence 200
    #playTone 1000 100 100
  }
}

this will get rid of the unwanted beep.
The normal local beep (proc send_roger_sound) in repeaterloci and simplexlogic will not be affected by this.
I believe this function does not serve a useful purpose and should be removed or at least be commented out by default.

Axel, DG1BBQ

UR3QJW, 2018-01-07:
I disabled this function for myself by default.

sm0svx, 2018-01-21:
The listen only issue is now fixed and also a new configuration variable, REMOTE_RGR_SOUND, has been added which can be used to enable and disable the feature. It defaults to off.

closed 2018-01-21 by sm0svx