sm0svx/svxlink #348: close short_identification during qso ?

f5swb, 2018-01-19:
Hi eveybody, 

I would like to know if it's possible to switch off the short identification during a qso on the air ?
I try FX_GAIN_LOW = -100 but only the first letter is heard not the next letters of the call ...
Any idea ?
many thanks in advance to all.
73's F5SWB/Dimitri.
svxlink 15.99.8 svxreflector.

f5vmr, 2018-01-19:
Hi Dimitri, in svxlink.conf simply make the short item time=0. However in F5ZGM-R the level of all fx is -15 while active, it can still be heard but only very faintly.

f5swb, 2018-01-19:
Many thanks chris for your quickly answer !

sm0svx, 2018-01-20:
@f5swb, are you saying that if you set FX_GAIN_LOW=-100, while someone is speaking you will still hear the first letter of the callsign but not the rest?

f5swb, 2018-01-20:
Yes my call is F5ZAU and I Heard only the F It's very strange isn't it ?

sm0svx, 2018-01-21:
Strange indeed... I'll mark this issue as a bug.

sm0svx, 2018-01-21:
@f5swb: It seems to work perfectly here. No part of the callsign is heard if I set FX_GAIN_LOW=-100. Tested with a standard RepeaterLogic.

You seem to be running a non-standard version of SvxLink. 15.99.8 have not been released by the SvxLink project. The latest release is 17.12.1. When running non-standard versions you need to contact the people who released that version or switch to using an official SvxLink release.

f5swb, 2018-01-25:
Hello sm0svx, in fact it works well with -100. 
Don't know why one time it didn't work ....

Many thanks.

closed 2018-01-25 by f5swb