sm0svx/svxlink #349: how to had temperature when the sytem execute a long identification

f5swb, 2018-01-21:
Hi everybody, I would like that the system announce the temperature (from Metarinfo i.e.) just after the hour :)

I think it will be here :

# Executed when a long identification (e.g. hourly) should be sent
#   hour    - The hour on which this identification occur
#   minute  - The hour on which this identification occur
#
proc send_long_ident {hour minute} {
  global mycall;
  global report_ctcss;
  global loaded_modules;
  global active_module;
  variable CFG_TYPE;

  spellWord $mycall;
  if {$CFG_TYPE == "Repeater"} {
    playMsg "Core" "repeater";
  }
  playSilence 500;

  if {$report_ctcss > 0} {
    playMsg "Core" "pl_is";
    playNumber $report_ctcss;
    playMsg "Core" "hz";
    playSilence 300;
  }

  playMsg "Core" "the_time_is";
  playSilence 100;
  playTime $hour $minute;
  playSilence 500;

    # Call the "status_report" function in all modules if no module is active
  if {$active_module == ""} {
    foreach module [split $loaded_modules " "] {
      set func "::";
      append func $module "::status_report";
      if {"[info procs $func]" ne ""} {
        $func;
      }
    }
  }

  playSilence 500;
}

The result could be : call / time / temperature is XX degrés 
Any idea to do this ?

Many thanks in advance.
73's F5SWB/dimitri.

n7ipb, 2018-01-22:
This is how I do it.

I have 1-wire temperature probes (this is on a raspberry pi) one of which is for the outside temperature.  It's checked once every five minutes and the current value is saved and the maximum and minimum temperatures since midnight are tracked.  These values are retrieved by SvxLink and passed to the MetarInfo module.

The code is placed just before where the active module status reports are called.
***
   set currentTemp [exec /usr/bin/gettemp]
   MetarInfo::temperature $currentTemp;
   set maxTemp [exec /usr/bin/get_max_temp];
   set minTemp [exec /usr/bin/get_min_temp];
   MetarInfo::rmk_minmaxtemp $maxTemp $minTemp;

 if {$active_module != ""} {
***

dl7ata, 2018-01-25:
This is my way to do it:
I chose between a nearby APRS-WX station or the next airport and metar data from http://www.aviationweather.gov .
In "namespace eval Logic { "-TCL, I add to procedure "long-ident":
--------------------------------------------------------------------------------------
 variable temp;                                                                                                                                             
  set warten 3                                                                                                                                               
  while "$warten > 2" {                                                                                                                           
    if {[catch {exec /bin/bash /home/svxlink/Wetter/Akt_Temp_holen.sh} result] == 0} {                                                                       
        if { $result == "tmready" } {                                                                                                                        
            set warten 1                                                                                                                                     
            source "/tmp/svx_akt_temp";                                                                                                                      
            playSilence 300;                                                                                                                                 
            playMsg "MetarInfo" "temperature";                                                                                                               
            playSilence 300;                                                                                                                                 
            if { int($temp) < 0} {                                                                                                                           
                playMsg "MetarInfo" "minus";                                                                                                                 
                set temp [string trimleft $temp "-"];                                                                                                        
            }                                                                                                                                                
            playNumber $temp;                                                                                                                                
            playMsg "MetarInfo" "unit_degrees";                                                                                                              
            playSilence 500;                                                                                                                                 
            break                                                                                                                                            
            }                                                                                                                                                
  } else {                                                                                                                                                   
        puts "BASH ERROR in Temperatur Ausgabe"                                                                                                              
        puts $result                                                                                                                                         
        set warten 1                                                                                                                                         
        break                                                                                                                                                
    }                                                                                                                                                        
   }                                                                                                                                                         
}  
-----------------------------------------------------------------------------------------------------
So grabbing a temperature of your choice is done in my bash script and the interface into svxlink is a small file which I read in that TCL-procedure. If you like, I can send it to you via PN or I can publish it in our german svxlink-forum in a separate topic.

Best
Frank
<DL7ATA>

PU5SMS-SAEZ, 2018-01-31:
Hi friends
Could anyone give me an idea of how to make the temperature and pressure transmission removed from the MetarInfo module to be transmitted every hour when the long identification is transmitted? ... I thought of a procedure that, by winning the time for long identification , invoke the MetarInfo to search the data of a single airport of my choice, and soon after, I would only read the data that interests me ... how can I implement this? ...

Thank you! 

Saez-PU3SMS / PU5SMS-BR.

f5swb, 2018-02-01:
Hello, I need exactly the same thing 😊

Please Help us to succeed !

dl7ata, 2018-02-26:
OK guys. Another simple way could be:

- create TCL-procedure temperature in /usr/share/svxlink/events.d/local and call it from procedure "long-ident". You can find the code here: [https://gist.github.com/dl7ata/c085a414d74ced0feb9e74b7dce3aac1](url)

- create a bash-script where you get the weather information from  [http://www.aviationweather.gov](url). You can find the code here: [https://gist.github.com/dl7ata/f70923d5977e0c26030dd85a963c2824](url)
- create a cronjob which runs every minute **before** ident-time like this: `29,59 * * * * /home/svxlink/getweather.sh > /dev/null`

f5swb, 2018-03-11:
Here's a simple method from F5NLG to get the meteo after the ident every 30 minutes ;)
1 - do a script in /etc/spotnik/ meteo.sh :
--------
#!/bin/bash
# read meteo
echo "*51#" > /tmp/svxlink_dtmf_ctrl_pty

2 - add this line in /etc/crontab
-----
30 * * * * root /etc/spotnik/meteo.sh

then restart the task /etc/init.d/cron restart

3 - modify MetarInfo.tcl in /usr/share/events.d to annonce what you want , I kept temperature , and QNH ( for the propagation) ;)

best regards.
F5SWB (F5ZAV - Réseau des Répéteurs Francophones).

closed 2018-03-11 by f5swb

PU5SMS-SAEZ, 2018-03-13:
Dear Frank, Thank you for your response but the links sent are broken for me. I did not get access to them. Would you have any other alternatives? I could visit the page in German language if you send the correct link. Please, if it is possible I would greatly appreciate your help.

Thank you very much.

73's PU3SMS/PU5SMS - SAEZ

PU5SMS-SAEZ, 2018-03-13:
F5SWB, thank you for sharing your solution. For me was the doubt of the directory ... / spotnik / ... Could you clarify what it is? ...

Thank you very much.
73's PU3SMS/PU5SMS - SAEZ

PU5SMS-SAEZ, 2018-03-13:
Dear Frank ... I ask you to disregard my previous message. In fact the link does not open directly, but copying and pasting worked ... Thanks again and sorry for the above comment.

73's PU3SMS/PU5SMS - SAEZ

f5swb, 2018-03-13:
yes it's true it's because I use a special distrib, for real svxlink users the directory should be /etc/svxlink :)

PU5SMS-SAEZ, 2018-03-14:
Okay, Dimitri ... I'm going to try your solution and then I'll put the results here in the forum ... Thank you, my friend.