sm0svx/svxlink #352: Possible issue with high channel count codec

Dloranger, 2018-01-28:
I am seeing something happen that I can't reconcile that hopefully will make sense to someone else.

I have gotten an OCTO8 soundcard from audioinjectors, which provides 8 channel of Tx an 6 channels of Rx audio.

Near as I can tell the board is working flawlessly, but when I try to run svxlink with even a single repeaterlogic channel the sql pin works only as long as the initial transmit (when svxlink first activates the channel).

at most once outside that window, I will get sql open/close/open and its gone from there, no more messages are printed acknowledging the sql pin has toggled

I have multiple windows looking at the system, sql pin gpio value does still toggle, the ptt pin does as well during the initial window, "top" shows the process at about 4-5% max after the initial tx burst, typically less than 1% but it seems to be doing something that moves the needle now and then.

I have disabled all my modules and my unofficial config variables have been working without issue on my 2 channel sound card previously so I don't suspect them, but open to revisiting if there is reason to be suspect.

any ideas to would cause this behavior like some internal assumption of 2 channels audio max per card, or something strange like that?  the log goes silent, so not much else I know to do for trying to narrow down the fault.

[svxlink.txt](https://github.com/sm0svx/svxlink/files/1671351/svxlink.txt)

sm0svx, 2018-01-30:
I have never tried using a multi-channel (>2) sound board but the intent of the code is to support multiple channels. One thing that may cause problems is the fact that there are not as many channels for RX as there are for TX. Don't know though, just guessing.

One thing you could try is to simplify it even further by just testing either RX or TX, not both at the same time. Just set the other audio device to another sound card or something like udp:127.0.0.1:10000. Not much help maybe but hopefully something useful can come out of it.

Dloranger, 2018-01-30:
Thanks for the ideas, I will give those a try when I get home.   This one had me stumped on where to go from here, the configs are down to a few basic settings on the 1 channel, all other logics are not activited.

I think I changed the number of channels to 6 in the config to help prevent potential confusion in the code, but I think its entirely possible that there is a driver issue lurking out there as well, its a fairly new design so i don't know how well tested the driver package is.

I will do more testing and see what I can come up with.

Dloranger, 2018-02-02:
made some progress on this 

when I use the Rx as UDP, it ignores the sql, when I use Tx as UDP sql works normally

[svxlink_6x_works.txt](https://github.com/sm0svx/svxlink/files/1689623/svxlink_6x_works.txt)
[svxlink_6x_hangs.txt](https://github.com/sm0svx/svxlink/files/1689624/svxlink_6x_hangs.txt)

Dloranger, 2018-02-02:
I have moved the asound files back to my known working for 2 channels, and the issue persists, so I think those files can be safely ignored

sm0svx, 2018-02-03:
Yes, I realized that you cannot use UDP for RX unless you actually put samples into the UDP socket. SvxLink is dependent on a steady flow of samples from the audio input device.

I don't quite understand if you got it to work when using UDP for TX. Did audio input and squelch work properly then? Any channel count?

Dloranger, 2018-02-03:
We played with this some more last night.

If we put the channel into simple mode, everything works as expected.

We then created a second channel of simplex and cross linked the channels.
This behaved the same as being in repeater mode.

In answer to your udp questions..

When playing with the udp, it was with a single channel only. Sql worked
fine with the tx as udp, but with the rx as udp, the sql pin is ignored.
The system still announces the id's on schedule in both scenarios.

<https://github.com/sm0svx/svxlink/issues/352#issuecomment-362797505>, or mute
the thread

Dloranger, 2018-02-07:
This issue may have self-resolved with new hardware.  I will close this issue if the problem remains gone after further testing.

Dloranger, 2018-02-10:
Unfortunately the problem remains.  I have hardwired the cards (pi and octo) together to make sure it wasn't a connector/cable.

I have also tried using VOX as the sql and this is also not working.

I can see that svxlink is interacting(read/write??) with the GPIO as when I monitor the I2C bus. I see traffic when svxlink is running, and it stops when I stop svxlink from running.  In the I2C traffic I can see the SQL bit changing so the pi is definitely seeing the hardware change states.

Is there a good place (TCL) to insert debug code to see what the sql portion of the code is doing/seeing?

sm0svx, 2018-02-10:
I'd guess that the squelch is not working because audio stops flowing somewhere. SvxLink is dependent on a constant flow of samples from the sound hardware while it's open for reading. Also sound output have to work so that all output samples are flushed because the squelch will not close until all samples have been flushed. The latter is a bit obscure but that's the way it's implemented at the moment. Since you got it to work with UDP audio output maybe it's a problem with that or maybe with the duplex functionality so that when you're using audio input and output at the same time something locks up.

I cannot say that there's a good place in TCL to put debugging code but you could try putting a printout in [Squelch::setSignalDetected](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/trx/Squelch.h#L347). Something like this:
```
std::cout << "### is_detected=" << is_detected << std::end;
```

Dloranger, 2018-02-10:
Interesting....

I probed the data signals from the audio card and the bit clock and L/R clock only toggle during a TX...

Looks like your guess was right.

Now to try to figure the sound card out to figure out why it is doing this.

Dloranger, 2018-02-11:
I got some further progress, but still having some issues.

I found out how to modify the drivers to send the clocks continuously, and with those changes I now have the sql detect working, but when SQL is toggled, I am not getting PTT, although it works during the IDs.

Playing short voice ID
Playing short CW ID
Tx1: Turning the transmitter ON
Rx1: The squelch is OPEN (8.20133)
Rx1: The squelch is CLOSED (8.09876)
Rx1: The squelch is OPEN (8.31067)
Tx1: Turning the transmitter OFF
Rx1: The squelch is CLOSED (8.16781)
Rx1: The squelch is OPEN (6.78721)
Rx1: The squelch is CLOSED (8.02196)
Rx1: The squelch is OPEN (6.80171)
Rx1: The squelch is CLOSED (6.8519)
Rx1: The squelch is OPEN (7.0156)
Rx1: The squelch is CLOSED (6.95125)
Rx1: The squelch is OPEN (6.91233)

sm0svx, 2018-02-11:
Does it decode any DTMF?

Dloranger, 2018-02-11:
I reverted back to an old config file and its working now, something in the new config file

Dloranger, 2018-03-02:
@sm0svx 

in some discussions around this issue and looking at the code changes for the new drivers, a question came up.

> https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L180-L182
> 
> get moved up to line 140

it appears the driver for this codec may be putting the device into standby mode of some sort.  i think the PCM allows for this.

I can't find in the code where alsa is configured, but am curious if svxlink checks the state of the soundcard and ensure it is in an active state as part of the initialization?

sm0svx, 2018-03-04:
I don't recognize that there are any special code for ALSA power management in SvxLink. All ALSA code is contained in [this file](https://github.com/sm0svx/svxlink/blob/master/src/async/audio/AsyncAudioDeviceAlsa.cpp) if you want to have a look at it.

Dloranger, 2018-03-09:
I am trying to understand this all better but I quickly get lost in the linkages as I read the documentation online.

What I see in the octo driver is a [switch](https://github.com/raspberrypi/linux/blob/rpi-4.9.y/sound/soc/bcm/audioinjector-octo-soundcard.c#L141) that seems to be depending on the state of the PCM device to set its clocks.  From what I am understanding, this is where the clock stopping issue is originating is in that the device is going to some type of stopped state when not playing out audio, which is not being accounted for.  

Purely from speculation since I don't understand this code at all...

its appears that there is some level of missing command at start-up to ensure the audio device is in a running state (like a "start" or "resume" command).  Why when the sound is playing it enters this state on its own and then exits when the playback stops.

This might be something unique to this board, but the raspbian maintainers said they didn't want to change the default behavior so to me that implies there is a good reason for having the various states exist.

Dloranger, 2018-04-08:
I have been discussing this a bit more with the sound card developer.  While I can force the clocking to stay on with the card (via a hacked driver), there is a new issue that is created in that synchronization gets lost so the playout channel (and assuming the record channel as well).

The syncronization is reset when the record/playout starts after the clocking has been stopped.  His suggestion is to somehow force one of the channels to always be recording or playing out from the instant when svxlink starts up.

Being that I have 6 record and 8 play channels, I could easily allocate the last play channel to playing out some dummy data,  is there is a way to force this type of playout in svxlink? any ideas on what might be doable to to try this configuration out?

Dloranger, 2019-01-09:
I have come to suspect this is all related to bug #96 after extensive testing with multiple types of sound cards from various sources.  They all suffer the same synchronization problem in the end.

closed 2019-04-17 by Dloranger