sm0svx/svxlink #353: Possible problem with AudioSelector

dl1hrc, 2018-01-29:
Hi Tobias,

I've running three logics (R,B,C) connected by two LINK's (LX, LY): R<->LX<->B and R<->LY<->C 
where R is a ReflectorLogic and B,C are "normal" LocalLogics

When started it's working perfectly but after some time it happens that AudioStream from the Reflector starts the normal repeater-functionality on LocalLogic's: TX(B)->is_on and TX(C)->is_on and the network audiostream is played over the LocalLogics. After some time the SQL's of the LocalLogics aren't closed any more (LocalLogic remain in TX=on mode) and the audiostream isn't played over it anymore.
The version before 5. Dec 2017 seems to be working, I've running v1.5.99.17-afsk_com at the moment.

When stopped in this state with Ctrl-X the following error message appears:

SIGINT received. Shutting down application...
svxlink:
/root/svxlink-sm0svx/svxlink/src/async/audio/AsyncAudioSelector.cpp:445:
void Async::AudioSelector::selectBranch(Async::AudioSelector::Branch*):
Assertion `(m_selected_branch == 0) || (m_selected_branch->streamState()
== STATE_IDLE) || (m_selected_branch->streamState() == STATE_WRITING)'
failed.

73s de Adi / DL1HRC

sm0svx, 2018-01-30:
Ok, I have never tried using two logics with the same ReflectorLogic. That should work of course if everything is functioning correctly.

If I understand you correctly the bug is possibly present in the 'maint' branch, which would raise the priority to fix it. Can you confirm this by reproducing the bug using a clean freshly compiled 'maint'? As usual a minimal config helps a lot.

dl1hrc, 2018-01-30:
I'm just installing the 17.12.1 (right?) and will look for the error with a reduced svxlink.conf-configuration. Unfortunately I'll not able to check the state very often tomorrow over the day since I will be on a short trip. Will give a feedback tomorrow evening.

sm0svx, 2018-01-30:
There are many ways. You can download the released zip and use that. If you normally build from git sources it may be easier to just do "git checkout maint". That will always give you the sources for the latest release.

dl1hrc, 2018-01-31:
The problem is pending but it seems to be self-healing. I checked out version 1.6.0.0 == 17.12.1 yesterday and installed it. Before I have deleted al libasync* libs manually and removed my locale tcl environment. Will compile all needed information...

dl1hrc, 2018-02-01:
Here is the svxlink.conf:
[GLOBAL]
MODULE_PATH=/usr/lib/svxlink
LOGICS=AsteriskLogic,RepeaterLogic,FMReflectorLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%d.%m.%Y %H:%M:%S.%f"
CARD_SAMPLE_RATE=48000
CARD_CHANNELS=1
LINKS=ReflectorLink,AsteriskNetLink

[FMReflectorLogic]
TYPE=Reflector
HOST=xx.xx.xx.xx
PORT=12345
AUTH_KEY="secret"
CALLSIGN="DB0HRC"

[AsteriskLogic]
TYPE=Repeater
RX=RxAsterisk
TX=TxAsterisk
OPEN_ON_SQL=40
OPEN_ON_DTMF=*
DEFAULT_LANG=de_DE
OPEN_SQL_FLANK=OPEN
SHORT_IDENT_INTERVAL=30
LONG_IDENT_INTERVAL=60
RGR_SOUND_DELAY=0
IDLE_TIMEOUT=10
NO_REPEAT=1
CALLSIGN=DB0HRC
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=de_DE

[RxAsterisk]
TYPE=Local
AUDIO_DEV=alsa:plughw:3,1,1
AUDIO_CHANNEL=0
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=2000
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
DEEMPHASIS=0
SQL_TAIL_ELIM=0
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=100
DTMF_SERIAL=/dev/ttyS0
DTMF_MAX_FWD_TWIST=18
DTMF_MAX_REV_TWIST=12
PREAMP=6

[TxAsterisk]
TYPE=Local
AUDIO_DEV=alsa:plughw:3,1,0
AUDIO_CHANNEL=0
PTT_TYPE=NONE
TIMEOUT=9000
TX_DELAY=0
PREEMPHASIS=0

[RepeaterLogic]
TYPE=Repeater
RX=Rx1
TX=Tx1
MODULES=ModuleHelp,ModuleParrot,ModuleTclVoiceMail
CALLSIGN=DB0HRC
SHORT_IDENT_INTERVAL=30
[... 133 lines cut ...]
31.01.2018 19:30:00.108: AsteriskLogic: Sending short identification...
31.01.2018 19:30:15.288: TxAsterisk: Turning the transmitter OFF
31.01.2018 19:30:15.313: Tx1: Turning the transmitter OFF
31.01.2018 19:45:10.067: FMReflectorLogic: Node left: DL7ATA
31.01.2018 19:45:13.942: FMReflectorLogic: Node joined: DL7ATA
31.01.2018 19:48:19.965: FMReflectorLogic: Talker start: DM0LEI
31.01.2018 19:48:19.969: TxAsterisk: Turning the transmitter ON
31.01.2018 19:48:19.970: Tx1: Turning the transmitter ON
31.01.2018 19:48:20.971: FMReflectorLogic: Talker stop: DM0LEI
31.01.2018 19:48:26.066: FMReflectorLogic: Talker start: DM0LEI
31.01.2018 19:48:28.413: FMReflectorLogic: Talker stop: DM0LEI
31.01.2018 19:54:43.484: RxAsterisk: The squelch is OPEN (93.3666)
31.01.2018 19:54:43.542: FMReflectorLogic: Talker start: DB0HRC
31.01.2018 19:54:45.661: FMReflectorLogic: Talker stop: DB0HRC
31.01.2018 19:54:45.661: RxAsterisk: The squelch is CLOSED (0)
31.01.2018 19:54:54.167: FMReflectorLogic: Talker start: DM0EA
31.01.2018 19:54:58.442: FMReflectorLogic: Talker stop: DM0EA
31.01.2018 19:55:00.537: FMReflectorLogic: Node left: DB0ERZ
31.01.2018 19:55:27.627: FMReflectorLogic: Node joined: DB0ERZ
31.01.2018 20:00:00.110: RepeaterLogic: Sending long identification...
31.01.2018 20:00:00.110: AsteriskLogic: Sending long identification...
**31.01.2018 20:00:18.830: TxAsterisk: Turning the transmitter OFF
31.01.2018 20:00:18.850: Tx1: Turning the transmitter OFF**

Between 31.01.2018 19:07:56.418 and 31.01.2018 19:16:51.975 the TX should switch from ON to OFF since the audio stream from the network has been stopped and IDLE_TIMEOUT is expired. But it doesn't. Later on at 31.01.2018 20:00:18.830 SvxLink resume normal functionality.

sm0svx, 2018-02-03:
Ok, thanks. I'll see what I can do about it.

dl1hrc, 2018-02-04:
I've added another logic and this caused the problem to occur faster. The error when stopping is the same I've provided:
/root/svxlink-sm0svx/maint/svxlink/src/async/audio/AsyncAudioSelector.cpp:445: void Async::AudioSelector::selectBranch(Async::AudioSelector::Branch*): Assertion `(m_selected_branch == 0) || (m_selected_branch->streamState() == STATE_IDLE) || (m_selected_branch->streamState() == STATE_WRITING)' failed.`

sm0svx, 2018-02-10:
Ok, thanks. I've been trying your config but cannot make it misbehave. I'll try on the Pi as well since that platform seem to provoke SvxLink into showing many strange bugs. I'll try to add another logic as well to see if the problem is more reproducible then.

The problem may be in the AudioSelector as you suggest but it's not for sure. I added some sanity checks in the AudioSelector previously and that is what is causing the assert. I'll definitely look into that though.

It's a bit hard to pinpoint this bug. What I normally do then is to use "git bisect" to find in which commit the bug was introduced. Since you have a known good commit git bisect is possible to use. Have you ever used it?

dl1hrc, 2018-02-10:
I don't know "git bisect", maybe it's easier for me to install older versions step by step to find the last working version. At the moment I've running v1.5.99.17 and the error is pending. My system is a "normal" mini-ITX board with Linux DB0HRC-neu 3.16.7-53-default #1 SMP Fri Dec 2 13:19:28 UTC 2016 (7b4a1f9) x86_64 x86_64 x86_64 GNU/Linux.
To raise the error I've running five logics here and it occurs between minutes.

sm0svx, 2018-02-10:
Ok. That's nice that you can reproduce it rather easily now!

You can try to step commit by commit if you like. The end result will be the same but learning git bisect may pay off since it's a very powerful way to find a bad commit.

What git bisect help you with is to step through a range of commits until you find the bad one. It does this using "interval halving" so you will find the bad commit in rather few steps if there are a lot of commits to search. Some times you can end up in a confusing state but many times it's very helpful. I wont explain the whole process here since it's better to just google it but these are the basic steps:

1. git bisect start \<bad commit\> \<good commit\>
2. Build and test
3a. If it's working do "git bisect good"
3b. If it's not working do "git bisect bad"
4. Git will automatically checkout the next commit to test
5. Repeat from item 2 until Git tells you which the bad commit is

sm0svx, 2019-07-15:
@dl1hrc, did you ever resolve this issue?