sm0svx/svxlink #356: logic SimplexLogic (invalid command name "}")

f5swb, 2018-02-04:
Hi, I've this fault since 2 days ago some have got an idea ?
** ERROR | Unable to handle event: SimplexLogic::manual_identification in logic SimplexLogic (invalid command name "}")
-- | --
I # the CTTSS announce in logic.tcl for the CTTSS 

#
# Executed when a long identification (e.g. hourly) should be sent
#   hour    - The hour on which this identification occur
#   minute  - The hour on which this identification occur
#
proc send_long_ident {hour minute} {
  global mycall;
  global report_ctcss;
  global loaded_modules;
  global active_module;
  variable CFG_TYPE;

  spellWord $mycall;
  if {$CFG_TYPE == "Repeater"} {
    playMsg "Core" "repeater";
  }
  playSilence 500;

  if {$report_ctcss > 0} {
#    playMsg "Core" "pl_is";
#    playNumber $report_ctcss;
#    playMsg "Core" "hz";
     playSilence 300;
  }

  playMsg "Core" "the_time_is";
  playSilence 100;
  playTime $hour $minute;
  playSilence 500;

    # Call the "status_report" function in all modules if no module is active
  if {$active_module == ""} {
    foreach module [split $loaded_modules " "] {
      set func "::";
      append func $module "::status_report";
      if {"[info procs $func]" ne ""} {
        $func;
      }
    }
  }

  playSilence 500;
}

Any idea ? 

Best regards.

F5SWB/ Dimitri.

Dloranger, 2018-02-04:
you have an extra close bracket above the playSilence 500 it looks like.

I suggest maintaining proper code indenting, it makes these much easier to spot.
[codeIndented.txt](https://github.com/sm0svx/svxlink/files/1693075/codeIndented.txt)

Dloranger, 2018-02-04:
can you post the whole file? I don't see anything wrong with what you posted

f5swb, 2018-02-04:
It's ok right now 

#  if {$report_ctcss > 0} {
#    playMsg "Core" "pl_is";
#    playNumber $report_ctcss;
#    playMsg "Core" "hz";
#    playSilence 300;
#  }
**}**
many thanks !

f5swb, 2018-02-04:
In fact I've got the problem again ,

send to you my logic.tcl

Dloranger, 2018-02-04:
looks like we are out of phase,

can you upload the txt file here as a file, I will take a look.

from the error, it looks like you have an extra close bracket somewhere. Happens to me regularly :-S

f5swb, 2018-02-04:
oups, ok I understand what you need ;)

[logi.tcl.txt](https://github.com/sm0svx/svxlink/files/1693119/logi.tcl.txt)

Many thanks

Dloranger, 2018-02-04:
looks like you missed commenting out a closing bracket on line 140

try this one
[logic.tcl.txt](https://github.com/sm0svx/svxlink/files/1693123/logic.tcl.txt)

f5swb, 2018-02-04:
it works fine right now many thanks .... any idea to add the temperature to the long idendification ;) ?

Have a good week !

best regards from france ! near strasbourg.
Dimitri.

Dloranger, 2018-02-04:
not sure where your source data for the temperature is, but if you can write the temperature to a file somewhere on a very slow interval (no point in wearing out the file system), you could then read in the file as text and use the "spellword" function

something like this <pseudo code>

set temperature [exec cat $absoluteDataPath]  #sample data "123C"
spellword $temperature

f5nlg, 2018-02-04:
an other way is to make a meteo.sh

of course MetarInfo shoud be loaded , configured ...

and dtmf_ctrl_pty also configured .

------------------------------------------------------

#!/bin/bash

# Lit la météo

echo "*51#" > /tmp/svxlink_dtmf_ctrl_pty

-------------------------------------------------------

then put the meteo.sh in cron for exemple at 30 minutes every hour

30 * * * * /etc/svxlink/meteo.sh

then restart cron

that's it :) in my exemple it have small mistake with airpot &cloud  
then temerature , dewpoint and qnh is read.

73" de Jean-Philippe / F5NLG

Modify MetarInfo.tcl like this :

###############################################################################
#
# MetarInfo module event handlers
#
###############################################################################

#
# This is the namespace in which all functions and variables below will 
exist.
# The name must match the configuration variable "NAME" in the
# [ModuleMetarInfo] section in the configuration file. The name may be 
changed
# but it must be changed in both places.
#
namespace eval MetarInfo {

#
# Check if this module is loaded in the current logic core
#
if {![info exists CFG_ID]} {
   return;
}

#
# Extract the module name from the current namespace
#
set module_name [namespace tail [namespace current]];

#
# An "overloaded" playMsg that eliminates the need to write the module name
# as the first argument.
#
proc playMsg {msg} {
   variable module_name;
   ::playMsg $module_name $msg;
}

#
# A convenience function for printing out information prefixed by the
# module name
#
proc printInfo {msg} {
   variable module_name;
   puts "$module_name: $msg";
}

#
# Executed when this module is being activated
#
proc activating_module {} {
   variable module_name;
   Module::activating_module $module_name;
}

#
# Executed when this module is being deactivated.
#
proc deactivating_module {} {
   variable module_name;
   Module::deactivating_module $module_name;
}

#
# Executed when the inactivity timeout for this module has expired.
#
proc timeout {} {
   variable module_name;
   Module::timeout $module_name;
}

#
# Executed when playing of the help message for this module has been 
requested.
#
proc play_help {} {
   variable module_name;
   Module::play_help $module_name;
}

#
# Executed when the state of this module should be reported on the radio
# channel. Typically this is done when a manual identification has been
# triggered by the user by sending a "*".
# This function will only be called if this module is active.
#
proc status_report {} {
   printInfo "status_report called...";
}

# temperature
proc temperature {temp} {
   playMsg "temperature";
   playSilence 100;
   if {$temp == "not"} {
     playMsg "not";
     playMsg "reported";
   } else {
     if { int($temp) < 0} {
        playMsg "minus";
        set temp [string trimleft $temp "-"];
     }
     playNumber $temp;
     if {int($temp) == 1} {
       playMsg "unit_degree";
     } else {
       playMsg "unit_degrees";
     }
     playSilence 100;
   }
   playSilence 200;
}

# dewpoint
proc dewpoint {dewpt} {
   playMsg "dewpoint";
   playSilence 100;
   if {$dewpt == "not"} {
     playMsg "not";
     playMsg "reported";
   } else {
     if { int($dewpt) < 0} {
        playMsg "minus";
        set dewpt [string trimleft $dewpt "-"];
     }
     playNumber $dewpt;
     if {int($dewpt) == 1} {
       playMsg "unit_degree";
     } else {
       playMsg "unit_degrees";
     }
     playSilence 100;
   }
   playSilence 200;
}

# sea level pressure
proc slp {slp} {
   playMsg "slp";
   playNumber $slp;
   playMsg "unit_hPa";
   playSilence 200;
}

# QNH
proc qnh {value} {
   playMsg "qnh";
   if {$value == 1000} {
      playNumber 1;
      playMsg "thousand";
   } else {
      spellNumber $value;
   }
   playMsg "unit_hPa";
   playSilence 200;
}

# output
proc say args {
   variable tsay;

   playSilence 100;
   foreach item $args {
     if [regexp {^(\d+)} $item] {
       sayNumber $item;
     } else {
       if {$item == "."} {
         playMsg "decimal";
       } else {
         playMsg $item;
       }
     }
     playSilence 100;
   }
   playSilence 200;
}

#
# Spell the specified number
#
proc playNr {number} {
   for {set i 0} {$i < [string length $number]} {set i [expr $i + 1]} {
     set ch [string index $number $i];
     if {$ch == "."} {
       playMsg "decimal";
     } else {
       playMsg "$ch";
     }
   }
}

# end of namespace
}

#
# This file has not been truncated
#

Dloranger, 2018-02-04:
Just a thought that struck me, you may not want to use the spellWord function, but rather one of the playNumber functions instead.  The reason is the numbers will be cleaner like forty-eight instead of four-eight. 

for the units, there may be another issue unless you create a new sound for the unit, as the default sounds and behavior is to use phonetic letters.  I would suggest rather using the word "degrees" if you can find it in the sounds, or you can create it with acapela-box.com tool online (almost free)

f5swb, 2018-02-05:
Hello, is there anyway to add the data of Metarinfo (I need only the temperature) just after the long identification ?

Many thanks for your help :)

closed 2018-02-05 by f5swb