sm0svx/svxlink #365: Idle Timer

dl1esk, 2018-02-15:
@sm0svx Tobias,

is it possible to add a Timer that measures the Time how long the Squelch was open and that just when Squelch open is longer than x Seconds the Idle Timer will be reset ?! That whould prevent that the Repeater stays open when some Playkiddies mean that they can hold the Repeater up when they push PTT short.
ATM the Timer got stopped and reset on every Squelch open. That whould be a nice feature that other Repeater Controllers already  have....

73 Sebastian

sm3sgp, 2018-02-15:
Have you tried the  SQL_FLAP_SUP_MIN_TIME and SQL_FLAP_SUP_MAX_COUNT?

dl1esk, 2018-02-15:
that can`t work...
u can reset the timer with one short Squelch open...
if u set the max count to 1 and u have mobile Stations, the Repeater will closed every time when the Squelch closes for short times...

sm0svx, 2018-02-18:
@dl1esk, maybe I don't understand but are you sure that the SQL_FLAP_SUP feature won't solve your problem?  It was added to suppress interferense that open the squelch very short a number of times in a row. Isn't that what you want?  That is, if someone is keeping the repeater open by continuously pushing the PTT quickly a number of times the repeater will close down.

dl1esk, 2018-02-18:
@sm0svx 
it was added to supress inteference right but when people push the PTT short just to hold the Repeater open thats no inteference. like idle timer starts...7 Seconds left ( from 15) short Squelch open, then the timer starts again with 15 seconds. The WX Steuerung and other repeater controllers have a function that u can set  that the idle timer only reset if the squelch is >x seconds open otherwise it will continue to countdown. That prevents that people play with the repeater..

sm0svx, 2018-02-18:
Ok, I understand. You can get similar functionality using the SQL_FLAP_SUP feature but not exactly the same. For example, if you set _MIN_TIME=1000 and _MAX_COUNT=5, the guy causing problems can reset the idle timeout at most five times using squelch openings that are less than a second long before the repeater will shut down. The biggest difference is that it will take longer for the repeater to go down.

dl1esk, 2018-02-18:
@sm0svx  Yes Tobias, but when u have Mobile Stations u have some SQL Flaps u can`t prevent this in mountainous areas thats why this function doesn`t work for that.

sm0svx, 2018-02-18:
Yes, I've seen it closing some times on mobile flutter and I guess that it's more of a problem in mountainous areas.

dl1esk, 2018-02-18:
@sm0svx maybe this is interesting for you: http://man7.org/linux/man-pages/man2/timer_create.2.html
there is a function that you can read the time left. before setting into idle maybe reading out that value store it into an variable and then measuring Sql open time. then the decision to feed the idle timer with the old value when sql is too short open or feed it with the config value if its open longer than x secs...

sm0svx, 2018-02-18:
SvxLink already have its own timer handling through the Async::Timer class.

dl1esk, 2018-02-18:
Yes... but u can only start a timer reset and delete a timer right?!
and u can`t read time left to expire right ?!

sm0svx, 2018-02-18:
It's solvable

dl1esk, 2018-02-18:
@sm0svx OK i spend a pot of Löfbergs Lila :-)

sm3sgp, 2018-02-18:
@dl1esk Tobias is from Gävle, there we drink Gevalia! ;-)

dl1esk, 2018-02-18:
@sm3sgp 
@sm0svx 
Ok Gevalia Dark for longer programming performance :-)