sm0svx/svxlink #370: MetarInfo and https with curl?

dl1hrc, 2018-02-26:
Hi Tobias,
aviationweather.goc has switched to https, so simple GET requests on port 80 are not answered anymore. A relatively simple way would be to use libcurl environment in ModuelMetarInfo.
Do you have an other/better idea?
73s de Adi / DL1HRC

sm0svx, 2018-02-26:
I don't have any experience with using https from a C++ application but libcurl definitely is the most known http client implementation so it probably is a good choice. It's also likely available on virtually all platforms so it's probably not problematic to add it as a dependency.

dl1hrc, 2018-02-26:
ok, thanks. Working on it right now. The only problem at the moment is to adapt the cmake scripts to use the needed curl libraries, especially the options for the linker. The correct operation of the tools (git, cmake, gdb,...) is often much more complicated than changing the whole programming concept ... for an amateur like me ;)
73s de Adi

f5vmr, 2018-03-10:
I have been doing some research on this issue and found that this site without HTTPS works.
http://tgftp.nws.noaa.gov/data/observations/metar/stations/LFLB.TXT with Limoges Bellegarde airport as an example. Would this site be more suitable to use? I've not any experience in the original code but I put this forward as a suggestion.

dl1hrc, 2018-03-10:
Thank you for the hint. You can use this web site as long as it is available (an example for tgftp.nws.noaa.gov is included in ModuleMetarInfo.conf) in plain http but more and more web sites switching to https.
Steve/DH1DM an me have continued the work on https support by using curl, a branch is available in my github space (branch metarssl). Maybe this extension can be used in other SvxLink parts too.

dl1hrc, 2018-03-15:
confirmed as working by Frank / DL7ATA and me. Many thanks to Steve / DH1DM for his work.
73s de Adi / DL1HRC

sm0svx, 2018-03-24:
Just to be clear, the code for this is in PR #375 but should probably be possible to merge into master soon.

dl1hrc, 2018-03-27:
correct, please check my branch 'metarssl'

sm3sgp, 2019-09-02:
Fix merged and available in 19.09.1 release.

closed 2019-09-02 by sm3sgp