sm0svx/svxlink #371: Segmentation Fault when trying to connect to Echolink Node

McPeebs, 2018-02-26:
Hi,

I am unable to connect to Echolink nodes. I am getting a segmentation fault and then SVXlink terminates. When I restart SVXlink, it gives me a Seg Fault error 2 to 3 more times, then on the fourth attempt the program will launch, but will still Seg fault when connecting to a Echolink node.

Anyone else had this issue? 

SimplexLogic: Macro command found: "EchoLink:9999#"
SimplexLogic: Activating module EchoLink...
DTMF command received in module EchoLink: 9999
Connecting to *ECHOTEST* (9999)
Tx1: Turning the transmitter ON
*ECHOTEST*: EchoLink QSO state changed to CONNECTING
Tx1: Turning the transmitter OFF
Segmentation fault

sm0svx, 2018-03-01:
Which version are you running?  Latest release (17.12.1) or latest git master?  On what hardware?  On what linux distribution?

sm3sgp, 2019-09-10:
@McPeebs Do you have more information on this issue?

closed 2022-05-03 by McPeebs