sm0svx/svxlink #381: Example tcl code for doing time based announcements

Dloranger, 2018-03-13:
Hoping someone can help me to locate some reference code I saw online a while back, but can no longer locate despite many attempts at recreating the right search patterns.

The code is for doing time based announcements where the file names include start and stop time stamps along with the file contents.

the basics of the code parse the file name for a date and time that the announcement will be activated at as well as when they will be terminated.

The idea for this code is to be able to schedule announcements far into the future that will last for a specific period of time and then auto expire.  this would be used for things like reminders about field day locations, other sponsored contests, arrl meetings, or the likes.

sm0svx, 2018-03-24:
Sounds like a nice idea. I have not seen any code for it though. Should probably be a standard feature in SvxLink.

n7ipb, 2018-03-25:
It sounds like what you want is a calendar module with regular and repeating events.  I've been just using cron but that's not really flexible enough.  

Rather than re-invent the wheel how about a module that interfaces to Google Calendar using it's API?

Dloranger, 2018-03-25:
I found where someone had already specifically built the code for svxlink,
so it would be very simple to bring in

sm0svx, 2018-03-25:
@n7ipb: Using Google calender wouldn't be my first choice since SvxLink should be able to operate without an Internet connection. It's an interesting idea though.

@Dloranger: Where?

Dloranger, 2018-03-25:
@tobias,

Thats the problem, I can't find it.  I thought I book marked it, but
apparently not.  If I could find it, I would be working on unifying it into
the code base.

I may start developing it again from scratch if I can't locate the code soon

dl7ata, 2018-03-31:
The non-official module "ModuleWeatherInfo.tcl" (proc is called  playAllWeatherInfos) contains a function to announce the time of a weather-warning file. This might be a basic function to settle on. Instead of using the filetime-stamp you can use a part of the filename and compare it with internal clock.
You can find it here: [svxlink.de/download/ModuleWeatherInfo.tcl](url) . Unfortunately some minor descriptions are in German language.

pe1chl, 2018-04-04:
It is of course "quite easy" to code this in TCL, as far as coding in TCL can be easy.
We have done a "propagation announcement" feature that operates indepently from the distributed
module, as in our repeater the EchoLink module is permanently active and so we cannot use other
modules.

It consists of a background program that prepares an audio file that is to be played, in our
case it is a loop that gets the propagation status from a website and copies the relevant
prepared message, in your case it could be a "cron" job that does the same.

The playout is done in our local version of Logic.tcl like this:
```
#
# Keep the start of transmission timestamp to detect short transmissions
#
variable start_of_transmission 0;
variable length_of_transmission 0;

# check if dx message is present, play it and remove it

proc dxmessage {} {
  set dxmsg "/dev/shm/gooddx.wav";
  set svxmsg "/dev/shm/svxlink.wav";

  if {[file exists $dxmsg] == 1} {
    puts "playing dx message $dxmsg";
    if {[file exists $svxmsg] == 1} {
      file delete $svxmsg;
    }
    file rename $dxmsg $svxmsg;
    playSilence 150;
    playFile $svxmsg;
    playSilence 150;
  }
}

# every minute, check if repeater is idle and if dxmessage is pending

proc dxmessage_timertick {} {
  variable start_of_transmission;

  if {$start_of_transmission == 0} {
    dxmessage;
  }
}

# Register the above timertick function
Logic::addTimerTickSubscriber "Logic::dxmessage_timertick";

proc send_rgr_sound {} {
... existing code ...
  dxmessage;
... existing code ...
}

#
# Executed each time the squelch is opened or closed
#   rx_id   - The ID of the RX that the squelch opened/closed on
#   is_open - Set to 1 if the squelch is open or 0 if it's closed
#
proc squelch_open {rx_id is_open} {
  variable sql_rx_id;
  variable start_of_transmission;
  variable length_of_transmission;

  #puts "The squelch is $is_open on RX $rx_id";

  if {$is_open} {
    set start_of_transmission [clock clicks -milliseconds];
    set length_of_transmission 0;
  } else {
    if { $start_of_transmission > 0 } {
      set length_of_transmission [expr [clock clicks -milliseconds] - $start_of_transmission];
    }
    if { $length_of_transmission > 1000 } {
      set sql_rx_id $rx_id;
    }
    set start_of_transmission 0;
  }
}
```

Dloranger, 2018-04-04:
I will take a look at both sets of code suggested.

I have done some modules already so I have already lost a lot time in some
of the tcl pitfalls.  Hopefully this one will come together nicely with
minimal head banging.

kj5fa, 2018-06-03:
Is this link still valid?
svxlink.de/download/ModuleWeatherInfo.tcl

Dloranger, 2018-06-04:
> Since the data protection basic regulation came into force in Europe, it
> is becoming increasingly difficult for small website operators to implement
> the legal requirements. To avoid warnings by lawyers, I have turned off
> this site svxlink.de until further notice. When the page comes back
> online, I can not say right now, sorry.
> 73s Adi / DL1HRC

would you be willing to share some sample code from the module to get this
started?

I am finally done with the swap meets and conventions for a while and
should be able to get back to this soon.