sm0svx/svxlink #384: Issue tracker being used as a discussion forum

sm0svx, 2018-03-24:
For those of you that does not use the mailing list:

I have noticed a disturbing trend that people have started to 
increasingly use the issue tracker on GitHub as a discussion forum. This 
is not what it's for. The issue tracker is for reporting issues, 
bugreports and feature requests.

The [mailing lists at SourceForge](https://sourceforge.net/p/svxlink/mailman/) have been the primary communication 
channel for SvxLink users since the start of the project. Maybe it's 
time for something new. I recently found [Groups.io](https://groups.io/) which may be a good 
candidate to replace the SourceForge mailing lists. It sort of is a combination 
of a forum and mailing list (and much more) which may suit more users. 
People can use email and/or the web interface to read and post.

I am evaluating it now. Please help testing it out by subscribing 
yourself to the group. It is very easy to register either while being 
logged in at the web user interface (https://groups.io/g/svxlink) or by 
sending an email to svxlink+subscribe@groups.io You can also register 
and post in one go by posting to svxlink@groups.io The post will not 
appear until a moderator have approved your first post.

If it all seems to work fine I'll probably move from SourceForge to 
Groups.io. See you there!

sm0svx, 2019-07-15:
Just pinging this issue to refresh it...

ea2abz, 2025-06-19:
Hello. I'm connecting Echolink to a conventional analog repeater. When my system completes the TX in VHF, it receives a "tail" from the repeater and adds that "tail" to the internet conference, creating an annoying ping-pong effect that causes the system to malfunction. Is there a way to configure the system to wait 2 seconds after the svxlink TX in VHF before connecting the RX, so it doesn't hear the tail of the system's own transmission to the repeater? I can't find a way.