sm0svx/svxlink #387: streaming don't succeed

f5swb, 2018-04-28:
Hello, 

I try to run the streaming according to [https://github.com/sm0svx/svxlink/wiki/StreamingAudioToInternet](url) but I can't succeed.
If I do all in the tutorial I lost my usb card :(
with a aplay - l : 

/etc$ aplay -l
**** Liste des Périphériques Matériels PLAYBACK ****
carte 0: Device [USB Audio Device], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 0: Device [USB Audio Device], périphérique 1: USB Audio [USB Audio #1]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
carte 1: Loopback [Loopback], périphérique 0: Loopback PCM [Loopback PCM]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 1: Loopback [Loopback], périphérique 1: Loopback PCM [Loopback PCM]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 2: ALSA [bcm2835 ALSA], périphérique 0: bcm2835 ALSA [bcm2835 ALSA]
  Sous-périphériques: 8/8
  Sous-périphérique #0: subdevice #0
  Sous-périphérique #1: subdevice #1
  Sous-périphérique #2: subdevice #2
  Sous-périphérique #3: subdevice #3
  Sous-périphérique #4: subdevice #4
  Sous-périphérique #5: subdevice #5
  Sous-périphérique #6: subdevice #6
  Sous-périphérique #7: subdevice #7
carte 2: ALSA [bcm2835 ALSA], périphérique 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0

Any help will be very appreciated for a novice like me ...

73's qro.

F5SWB. Dimitri.

pe1chl, 2018-05-30:
Always refer to audio devices using their NAME, not a NUMBER, because the number can and will
change when you make changes to the system like adding the Loopback device.
Do: cat /proc/asound/cards
to find the names of your cards. This is what is between the square brackets.
So, do not use things like this:
AUDIO_DEV=alsa:plughw:0
But use:
AUDIO_DEV=alsa:plughw:bcm2832
(for example)

f5swb, 2018-06-04:
Many thanks

closed 2018-06-04 by f5swb