sm0svx/svxlink #389: Script eventsource not working

dterweij, 2018-05-10:
Hello,

The script eventsource.pl is not working out of the box,
The index.html files refers to pictures and javascript script (jQuery) that are not included.
The example is too much.  It needs a basic example just with one  Rx and one Tx.
It needs to describe what other variables you need to change. The README is unclear and missing things.
How is it working?  What is the eventsource.pl doing? How the data gets displayed on the browser?
What are the urls about in the Repeater info json ?
If my Rx/Tx is named:  RxRadio/TxRadio  in svxlink.conf, where put that in eventsource? or is that all automagicly discovered?
Why is that ,htaccess file? What is it doing?

Anyway, it is not working with given documentation and files. After playing with it for a day I give up and try again when all is proper documented and all missing items are added.

A example needs to be simple and working :)

pe1chl, 2018-05-13:
It requires some expertise in Linux system management and the ability to read a terse README.
E.g. the way to install jQuery is right in there.  Apparently you missed it.
Sorry but the svxlink project is not a course in Linux system management.

dterweij, 2018-05-13:
Then remove this from this project. I am expirenced with Linux and coding. The JSON response on port 1535 at the eventsource.pl deamon is not a full valid JSON response.

The deamon is running fine but the www pages are doing nothing here.  No bars and info.

I am writing now my own deamon.   Injecting all the PTY data into mysql database.  Then I can use jQuery/Ajax to poll the database.  Thereby I also solve the JSON cross domain problem.

Just like I said,  remove this from the project.  It is not working. And to debug why is time consuming.

sm3sgp, 2018-05-20:
@dterweij 
Sounds interesting with your daemon! If possible, can you share your setup so I can give it a try. Do you have an instruction?. We already have working setup with the eventsource.pl here, but would be nice to test something new.

dterweij, 2018-05-27:
@sm3sgp Busy writing it.  Injecting into the (remote) database is ready.  Now to see what the best injecting method is. For now it injects every single receiver line.  Each row, with fields rx, sql, lvl (no date/time yet).
Used only pty reading part from Rob's eventsource.pl. I mail you when i have something that works :)

dterweij, 2018-05-27:
It seems sysread() perl function is too slow. It skips lines that are send to the STATE_PTY file from svxlink. Or it is because the MySQL calls that slows it down a bit and the sysread() just does not see new lines. sysread() is also not buffered and it is mentioned as a slow function.  So a little setback. I try first other file functions to see  if it goes faster. If not working, I step away from perl and see what other language i can use to try :)

dterweij, 2018-05-27:
Follow the progress here https://github.com/dterweij/pty2sql  so this issue stays clean :)  Also I close it.

closed 2018-05-27 by dterweij