sm0svx/svxlink #390: Does not switch TX RX the system SVXLINK / Raspberry PI3 B

catsilver, 2018-05-11:
Does not switch TX RX the svxlink system, tested on a raspberry with a digitus usb to serial adapter, on windows it works perfectly, for use radio link ?

the audio comes out perfectly....... I hope to solve this problem Thanks

AKGhost, 2018-05-13:
I have a TK-705 simplex link and it works fine with a USB Serial. Mine is a prolific.
I trigger both RTS and DTR with my setup and no issues.

catsilver, 2018-05-13:
I use a digitus adapter, but it does not go in tx and rx? I have done all the settings via svxlink.conf etc. etc.

but fine on windows 8.1 pro

catsilver, 2018-05-14:
I using the radio Icom ic 2820, interface....

catsilver, 2018-05-17:
I using the 14. and 17.12.1......very bug problem?

catsilver, 2018-11-22:
If you could kindly have the svxlink image file with the settings? many thanks.....

iw0red, 2018-11-22:
Hi Fabio, check www.svxlink.de if you don’t find out email me
marco.brunozzi@gmail.com.

73 Marco IW0RED

http://iw0.red

catsilver, 2018-11-24:
> Hi Fabio, check [www.svxlink.de](http://www.svxlink.de) if you don’t find out email me marco.brunozzi@gmail.com. 73 Marco IW0RED http://iw0.red
> -- Sent from Gmail Mobile

I downloaded the file, I tried it on my raspberry ..... but it does not start !!!!!!!!

sm0svx, 2019-07-16:
Please use the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/svxlink-devel) or [groups.io](https://groups.io/g/svxlink) to ask questions.

closed 2019-07-16 by sm0svx