sm0svx/svxlink #396: GPIO Setup Script Bug

alleghenycreative, 2018-06-15:
There is a typo in /usr/sbin/svxlink_gpio_up 

In the script on line 37 the variable for path is missing underscore: $GPIOPATH…should be $GPIO_PATH

iw0red, 2018-06-15:
Hi,
can this bug cause a trouble?
I found out on my orange pi zero. I can tried to use GPIO like GPIO_IN_HIGH. Svxlink start to work but if the GPIO goes up it never goes down.

RepeaterLogic: Event handler script successfully loaded.
Rx1: The squelch is OPEN (0)
Tx1: Turning the transmitter ON
.....

I checked the GPIO with command gpio readall it works fine.

73 Marco IW0RED

alleghenycreative, 2018-07-01:
I am not looking at it now, but seam to recall that it had something to do with setting the permission of the pin. So the pin looks like it is setup, but permissions are not set right for the svxlink user to be able to access it. So it would work properly when accessing the pin directly via /sys/class/gpio as a root user, but the svxlink user would permission issues that are not made apparent. And this all has to do with a missing underscore in a variable.

sm0svx, 2019-07-16:
Fixed a while back

closed 2019-07-16 by sm0svx