sm0svx/svxlink #397: temperature in french ?

f5swb, 2018-06-23:
Hello, I would like to know how it's possible with the #51 metar info to get the temperature as : 
twenthytwo degre instead of two two degre as the time announcement ?

Tks for a reply.

73's F5SWB.

f5nlg, 2018-06-23:
Hello

as i said , you have to install local.tcl for french language , and make 
some adjustment on metarinfo.tcl and tries wich gave the better diction. 
you have also to configure your pi in french and not in english ...

you can found it on spotnik it works for all platform ... with best 
diction i have found .

73" F5NLG

sm0svx, 2019-07-16:
Please use the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/svxlink-devel) or [groups.io](https://groups.io/g/svxlink) to ask questions.

closed 2019-07-16 by sm0svx