sm0svx/svxlink #399: SIGILL, Illegal instruction - crash while stating as daemon

dl7ata, 2018-07-04:
I got a crash when starting svxlink v1.6.99.10 as usually with /usr/bin/svxlink --daemon --logfile=/var/log/svxlink without any further message in logfile.

When i start svxlink in GDB I got this output:
Starting program: /usr/bin/svxlink 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Cannot access memory at address 0x0

Program received signal SIGILL, Illegal instruction.
0xb679fe88 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0

In GDB I can continue with "C" and svxlink is running in terminalmode. But as daemon, it still crashes, I guess caused by the above error.

dl7ata, 2018-07-05:
Last night I got another crash while connecting into the svxreflector:

> Logic_Ostlink: Authentication OK
Logic_Ostlink: Connected nodes: DB0DMO, DB0MGN, DO0KGS, DB0EEE, DB0TDR, DB0SVX, DO0RM, DB0ERZ, DM0LEI, DB0EDF, DB0MUE, DB0SLF, DL2HXA, Zello_Gateway, DB0FTS, DM0EA, DB0PER, DB0TGO
Logic_Ostlink: Using audio codec "OPUS"
Logic_DL7ATA: Authentication OK
Logic_DL7ATA: Connected nodes: DL5XL, DB0MGN-2m, DD6LK, DO0SE, DL6JOG, DB0FTS, DB0TGO
Logic_DL7ATA: Using audio codec "OPUS"
*** WARNING: Disconnected from APRS server
*** WARNING: Trying to reconnect to APRS server
[New Thread 0xb3741430 (LWP 29270)]
[Thread 0xb3741430 (LWP 29270) exited]
Connected to APRS server 129.15.108.112 on port 14580
*** ERROR: The directory server closed the connection before all data was received

EchoLink directory status changed to ?
svxlink: /home/svxlink/svxlink/src/async/core/AsyncTimer.cpp:150: void Async::Timer::setEnable(bool): Assertion `(m_timeout_ms >= 0) || !do_enable' failed.

Program received signal SIGABRT, Aborted.
0xb69ad8e6 in ?? () from /lib/arm-linux-gnueabihf/libc.so.6
<

dl7ata, 2018-07-08:
I got the reason:
When i received my repeatercallsign, I changed it in svxlink.conf/locationinfo. Old version, running without any error:
`[LocationInfo]                                                                                                                                               
APRS_SERVER_LIST=rotate.aprs.net:14580                                                                                                                       
STATUS_SERVER_LIST=aprs.echolink.org:5199                                                                                                                    
LON_POSITION=13.13.59E                                                                                                                                       
LAT_POSITION=52.34.16N                                                                                                                                       
CALLSIGN=EL-DL7ATA                                                                                                                                           
FREQUENCY=430.025                                                                                                                                            
TX_POWER=1                                                                                                                                                   
ANTENNA_GAIN=13                                                                                                                                              
ANTENNA_HEIGHT=14m                                                                                                                                           
ANTENNA_DIR=-1                                                                                                                                               
PATH=WIDE1-1,WIDE2-1                                                                                                                                         
BEACON_INTERVAL=60                                                                                                                                           
#TONE=136                                                                                                                                                    
COMMENT=| EL-Node#41041 | solar powered since 7/2016 | DL7ATA@Winlink.org | svxlink & http://elenata.datenport.net |`
New part with seg-fault:
`[LocationInfo]                                                                                                                                               
APRS_SERVER_LIST=rotate.aprs.net:14580                                                                                                                       
STATUS_SERVER_LIST=aprs.echolink.org:5199                                                                                                                    
LON_POSITION=13.13.59E                                                                                                                                       
LAT_POSITION=52.34.16N                                                                                                                                       
CALLSIGN=EL-DB0TGO                                                                                                                                           
FREQUENCY=430.025                                                                                                                                            
TX_POWER=1                                                                                                                                                   
ANTENNA_GAIN=13                                                                                                                                              
ANTENNA_HEIGHT=14m                                                                                                                                           
ANTENNA_DIR=-1                                                                                                                                               
PATH=DB0TGO-15,WIDE2-1                                                                                                                                       
BEACON_INTERVAL=30                                                                                                                                           
#TONE=136                                                                                                                                                    
COMMENT=| EL-Node#41041 | solar powered since 7/2016 | sysop: DL7ATA@Winlink.org | svxlink & http://elenata.datenport.net |`
So I assume that the digi-path caused the crash.

sm0svx, 2019-07-16:
@dl1hrc @dl7ata, is this fixed? I don't remember. If it isn't, reopen this issue.

closed 2019-07-16 by sm0svx