sm0svx/svxlink #401: priority function svxserver

dl1esk, 2018-07-16:
Hi Tobias and everyone here,

@sm0svx is it possible to add a priority function for connected Repeaters together over the Svxserver ?!
Maybe configurable in the conf file from the server, to decide priority yes or no.
at the moment local HF on the Repeater has always Pritority but in some circumstances it`s better to change this...

73 Sebastian DL1ESK

dl1esk, 2019-07-15:
i wanna push this upwards because i mean this is an important function what we need.
i already made a fix for my repeater and i gave the HF way the same priority as the Modules way so EL can`t be eliminated when somebody tries to disturb via HF . Logic.cpp Line 478 
tx_audio_selector->enableAutoSelect(audio_from_module_idle_det, 0); 
to
tx_audio_selector->enableAutoSelect(audio_from_module_idle_det, 30);
and Line 484
logic_con_out->enableAutoSelect(passthrough, 0);
to
logic_con_out->enableAutoSelect(passthrough, 30);

maybe somebody has an Idea how to implement a Config Value to select the priorities...

73 Sebastian

dl1esk, 2019-07-16:
@sm0svx pse have a look at it

sm0svx, 2019-07-16:
Noted