sm0svx/svxlink #404: Could NOT find Threads (missing: Threads_FOUND)

jetski28, 2018-07-26:
I am trying to compile the latest source code on Raspbian Stretch as there is an update that I require to invert SQL on a HIDRAW device that is not available in the latest build but it keeps failing at the following...

root@raspberrypi:/opt/svxlink/src/build# cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc -DLOCAL_STATE_DIR=/var -DUSE_OSS=NO -DUSE_QT=NO ..
-- Reading versions file...
-- SvxLink user = svxlink
-- SvxLink group = svxlink
-- Could NOT find Threads (missing:  Threads_FOUND)
-- Could NOT find Threads (missing:  Threads_FOUND)
-- With Systemd = OFF
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindThreads.cmake:212 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  svxlink/trx/CMakeLists.txt:52 (find_package)

Assuming I am on the right page, pThreads appears to be installed on here but CMake is not seeing it. I have installed all the required libraries as outlined in the wiki's but having no luck. Any ideas where I am going wrong?

pe1chl, 2018-09-08:
You need to install the "-dev" versions of the library packages.

sm0svx, 2019-07-16:
Please use the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/svxlink-devel) or [groups.io](https://groups.io/g/svxlink) to ask questions.

closed 2019-07-16 by sm0svx