sm0svx/svxlink #409: Problem with Svxlink starting on reboot after upgrading to Ubuntu 18.04

kb5zo, 2018-08-17:
I had Svxlink 17.12.2 running on Ubuntu 16.04 without any problems but when I upgraded to Ubuntu 18.04 when I reboot the computer the Svxlink modules do not respond to radio commands until an Echolink connection is made then normal radio commands function like MetarInfo, Parrot, Help, and Echolink. When the Echolink connection is made the announcement is not transmitted on the repeater. Echolink audio is transmitted and radio audio is sent to the Echolink user OK. The disconnect announcement is sent on the repeater.
 The log file shows connection to the server but does not show TX or RX or DTMF decoded characters until the Echolink connection is made. I did have to reinstall TCL8.6 because the upgrade removed it. I have checked to see if any of the other packages were missing and also tried a reinstall of Svxlink with no change. Any suggestions to try next?

pe1chl, 2018-09-08:
After such an upgrade you should re-compile the package.  Remove the build directory before, so all library and package versions are re-detected.
Do not run a binary package from an (older or newer) OS version.  It may work, but it may also cause issues like you mention.
Unfortunately, when you have done different experiments including installing packages and compiling your own, there may now be several incompatible library versions lingering around.  You need to remove those first.  Start by uninstalling any installed packages.

kb5zo, 2018-09-08:
Since my first post I have tried the following:
Installed 18.04 64 bit from scratch and 17.11.2 from scratch from what I downloaded.
Installed 18.04 32 bit from scratch and 17.11.2 from scratch.
Installed 16.04.5 64 bit from scratch and 17.11.2 from scratch.

All 3 act the same. Using Crontab to launch Svxlink it launches ok but untill someone connects via Echolink the program does not seem to be listening to the internal audio card or send audio. When they connect no connect id is sent although the repeater keys up, Once the echolink user sends their first transmission everything works normal.

I installed 16.04.5 32 bit from scratch and 17.11.2 from scratch. 
This worked properly with no problems that I noticed. I will stay with this release as it does work.

It looks like there is something that has changed in 16.04 64 bit and both 18.04 versions.
Svxlink 17.11.2 was downloaded on May 8 2018.
The Ubuntu versions used the net install download which was retrieved about 15 minutes before the install over the past several weeks. The original test was done on the day Ubuntu released their system release upgrade.
The computer is an HP DC7800 using its internal sound device. 
I also noticed that the 18.04 repository had Svxlink-server 17.11 in it. The 16.04 64 bit had the Svxlink-server 15 version and the 32 bit has the Svxlink-server 14 version in it.

sm0svx, 2019-07-16:
This does not sound like a SvxLink bug so I'm closing this issue. Try installing the latest Ubuntu and install the included SvxLink packages. A systemd start script should be included so starting SvxLink at boot should be as simple as systemctl enable svxlink. No cron magic needed.

Please reopen if you still think that this is a bug.

closed 2019-07-16 by sm0svx