sm0svx/svxlink #412: Module Echolink: AUTOCON_ECHOLINK_ID fail on several ID on Raspberry Pi

mrgadotti, 2018-08-29:
Scenario:
- RASPBIAN STRETCH LITE Version:June 2018 Release date:2018-06-27
- Raspberry Pi using ethernet cable or wireless interface
- Trying different logged calsing 
- Trying using different proxy 
- Trying using opened UDP ports 5198 and 5199
- Installed packages:
    svxlink-calibration-tools 17.12.2-1
    svxlink-gpio 17.12.2-1
    svxlink-server 17.12.2-1

Log from svxlink:
Wed Aug 29 11:28:28 2018: EchoLink directory status changed to ON
Wed Aug 29 11:28:31 2018: --- EchoLink directory server message: ---
Wed Aug 29 11:28:31 2018: EchoLink Server v2.5.9997
Wed Aug 29 11:28:31 2018: 
Wed Aug 29 11:28:31 2018: ECHOEC2-1: Herndon, VA USA
Wed Aug 29 11:28:31 2018: 
Wed Aug 29 11:28:40 2018: SimplexLogic: Activating module EchoLink...
Wed Aug 29 11:28:40 2018: ModuleEchoLink: Trying autoconnect to 9999
Wed Aug 29 11:28:40 2018: Connecting to *ECHOTEST* (9999)
Wed Aug 29 11:28:55 2018: ModuleEchoLink: Trying autoconnect to 9999
Wed Aug 29 11:28:55 2018: Connecting to *ECHOTEST* (9999)

Checking the internet traffic on Linux, I have been noted that there aren't traffic when trying to connect on ID, using ECHOTEST or conference ID. The same issue happens  using different callsign and different network interface.

sm0svx, 2019-07-16:
@mrgadotti, is this a bug?

mrgadotti, 2019-07-20:
I've change Logic.tcl to the default values and it solve the problem. I think that I had configured a wrong parameter.

closed 2019-07-20 by mrgadotti