sm0svx/svxlink #414: Incoming UDP packet has the wrong source UDP port.

iw0red, 2018-09-01:
Hi everyone,
frequently on the log I read this:

01.09.2018 22:03:29: IR0UGM: Login OK from 185.78.18.7:35524 with protocol version 1.0
01.09.2018 22:06:29: *** WARNING[IR0UGM]: Incoming UDP packet has the wrong source UDP port number
01.09.2018 22:06:44: *** WARNING[IR0UGM]: Incoming UDP packet has the wrong source UDP port number
01.09.2018 22:06:59: *** WARNING[IR0UGM]: Incoming UDP packet has the wrong source UDP port number
01.09.2018 22:07:14: IR0UGM: disconnected: Connection closed by remote peer

What's wrong?
73 Marco IW0RED

iw0red, 2018-09-01:
On the client site:
ReflectorLogic: Connection established to 212.43.96.84:5300
ReflectorLogic: Authentication OK
ReflectorLogic: Connected nodes: EchoLink, IR0UAS, IR0UAT, IR3DU, IR6UN, IR6UDJ, IR6UZ, IR5UK, IR0BP-430
ReflectorLogic: Using audio codec "OPUS"
Tx1: Turning the transmitter OFF
ReflectorLogic: UDP Heartbeat timeout
ReflectorLogic: Disconnected from 212.43.96.84:5300: Locally ordered disconnect
ReflectorLogic: Connecting to svxlink.iw0red.it:5300
ReflectorLogic: Connection established to 212.43.96.84:5300
ReflectorLogic: Authentication OK
ReflectorLogic: Connected nodes: EchoLink, IR0UAS, IR0UAT, IR3DU, IR6UN, IR6UDJ, IR6UZ, IR5UK, IR0BP-430
ReflectorLogic: Using audio codec "OPUS"

......

pe1chl, 2018-09-08:
Probably a NAT router...
I would recommend setting up a VPN.
We run everything on AMPRnet (HAMNET) addresses with VPN where required when only internet available.

iw0red, 2018-09-09:
Thanks for the answer. 
Sound strange. There are a lot of device on the same network/router like dstar c4fm dmr etc. I think it's a svxlink issue. I don't know the svxlink protocol and I'm not a programmer but I made a lot of test.
73 Marco IW0RED.

iw0red, 2018-09-10:
The creator don't answer :-)

pe1chl, 2018-09-10:
It is not a commercially supported product, you can report problems but usually nothing is done unless you do it yourself or it is a really important problem that affects everyone.  Also this probably isn't really a bug but only the effect of your network layout and I provided you with 2 possible workarounds.  It is your decision if you want to go with those.

iw0red, 2018-09-10:
Funny :-)

iw0red, 2018-09-10:
You need a vpn....
(you: I would recommend setting up a VPN.)

iw0red, 2018-09-10:
I forgot: 
thanks for the answer. You are the only one. :-)
Sorry for my english.
73 Marco IW0RED

RichNeese, 2019-06-26:
CLOSE

sm0svx, 2019-07-15:
@iw0red, hope you got it solved. It's a security feature that SvxLink/Reflector will only accept UDP packets originating from the same IP address as the accompanying TCP connection originates from. If the IP address changes now and then you will get this kind of problem. A VPN may be the only way to solve the problem given the current implementation.

closed 2019-07-15 by sm0svx