sm0svx/svxlink #417: feature request, multiple CTCSS on RX

riku2015, 2018-09-16:
Is there any way adding a feature where there can be RX same time multiple different CTCSS? 
Good example here on Finland we have a roaming RX-CTCSS what is common on repeaters just in case there is someone driving around the country and he can use it temporary on multiple repeater as moving on. Some info about it https://automatic.sral.fi/?p=ctcss

other where it is handy on radios where ctcss is modified and not possible to add exact one, so there is other option for it.

I know it can be done now with rxvoter but it will just increase a delay way too much.

best 73, OH1E

sm0svx, 2019-07-16:
There are plans for this.

sm3sgp, 2019-09-01:
@riku2015 This is now possible in the experimental reflector_tg branch.

CTCSS_FQ
If CTCSS (PL,subtone) squelch is used (SQL_DET is set to CTCSS), this config
variable sets the frequency of the tone to use. The tone frequency ranges from
67.0 to 254.1 Hz. There actually is nothing that will stop you from setting the
frequency to something outside this range but there is no guarantee that it will
work.

If desired it is possible to specify a space separated list of tone frequencies
in which case the squelch will open on all the tone frequencies.

riku2015, 2019-09-01:
nice to hear, i assume it is released under 19.09 ? will test it out soon, thanks very much.

sm3sgp, 2019-09-02:
No, this feature is not the release 19.09. It is currently in the reflector_tg branch but migth be merged into master soon.

closed 2025-03-18 by sm3sgp