sm0svx/svxlink #420: PTT_PTY link issue

M0ORY, 2018-09-25:
Hello. Not sure if I've found a bug here or it's just me. I'm trying to set up svxlink to key my icom ic7100 which can't be keyed on DTR / RTS without building a hardware interface so I went down the PTY route an wrote a little script to get hamlib rigctl to key the radio for me. The script is working fine and I've set up a PTY which is on /dev/pty/5 which I can connect to with screen and send the T/R commands and key the radio just fine. I added the following to svxlink.conf

```
PTT_TYPE=PTY
PTT_PTY=/dev/pts/5
```
Now when I try to start svxlink it tries to symlink /dev/pts/5 to /dev/pts/7 and this fails and the server shuts down. Although the directory /dev/pts appears to be writable I can't as root create any links or even touch a file in there. This is running on Ubuntu 18.04 with the repo version of svxlink

```
root@binky-Aspire-S3:~# svxlink
SvxLink v1.6.0 Copyright (C) 2003-2017 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 48000Hz

Starting logic: SimplexLogic
Loading RX: Rx1
Loading TX: Tx1
*** ERROR: Failed to create PTY slave symlink /dev/pts/4 -> /dev/pts/5
*** ERROR: Could not initialize TX "Tx1"
*** ERROR: Could not initialize Logic object "SimplexLogic". Skipping...
*** ERROR: No logics available. Bailing out...
```
I'm not sure if this is just me missing the point but I'm not sure why it would need to create a link.
I searched high and low and couldn't find any other references to this issue so thought I'd open a issue just in case I have indeed found a bug.

M0ORY, 2018-09-26:
Yes I was being stupid. For anyone else making the same mistakes as me here's what I did wrong and what I should have been doing.

I had assumed that PTT_PTY was expecting to connect to one end of EXISTING PTY and thus would need to be set to /dev/pts/# This is not the case and svxlink creates it's own PTY. One should specify a path to a directory to place a file which will then be symlinked the the /dev/pts/# created by svxlink.

If this still doesn't make sense Here's what I did that worked step by step
I created an empty directory. in my case /tmp/svxlink
I configured the PTT_PTY in the tx1 part of svxlink.conf to point to /tmp/svxlink/ptt
Started svxlink
amended my script that calls rigctl to key radio to read a character at a time from /tmp/svxlink/ptt
Here's my script
```
#!/bin/bash

while IFS= read -r -n1 char
do
if [ $char == "T" ]
  then
    `rigctl -m 370 -r /dev/ttyUSB0 -s 19200 T 1`
fi
if [ $char == "R" ]
  then
    `rigctl -m 370 -r /dev/ttyUSB0 -s 19200 T 0`
fi

done < "${1:-/tmp/svxlink/ptt}"
```
Run the script (it just sits there listening for input until you kill it) connect to your node and press the transmit button. your rig will key up. 

Now to see if I can do the same on rx for squelch detection some how. Hopefully this helps someone in future.

sm0svx, 2019-07-16:
If you can find the time, write up a wiki page about it on the SvxLink wiki [ConfigExamples](https://github.com/sm0svx/svxlink/wiki/ConfigExamples).

closed 2019-07-16 by sm0svx