sm0svx/svxlink #422: QTel not starting...

KI5JYD, 2018-09-30:
It worked yesterday... now it won't start.

In a terminal, I get these three errors...
```
bind: Cannot assign requested address
bind: Cannot assign requested address
Could not initialize network listen ports
```

I have searched the issues and this is a new one. I'm at wit's end. I would normally reinstall the application (which I did by `sudo apt-get -y purge qtel && sudo apt-get -y install qtel`), which did nothing. I would also try to remove the local conf files but I can't find them in `/home`. Nothing worked.

I'm sorry if this is vague. I have no idea what to do and this is the only output I have and **please** don't tell me to use a VPN or bug my ISP to give me a static IP (the latter option, my ISP told me that is only valid for business customers... so that is **not** an option).

KI5JYD, 2018-09-30:
Nevermind....

I figured it out.

IPv6 is the culprit. I had a kernel update and it automatically enabled IPv6 while the previous kernel had it disabled. I hate the Ubuntu team for fscking up stuff when it's not needed.

closed 2018-09-30 by KI5JYD

hristo-stoyanov, 2020-05-09:
> Nevermind....
> 
> I figured it out.
> 
> IPv6 is the culprit. I had a kernel update and it automatically enabled IPv6 while the previous kernel had it disabled. I hate the Ubuntu team for fscking up stuff when it's not needed.

How u solved problem ?