sm0svx/svxlink #425: Close EchoLink connection on killall?

DrW0if, 2018-10-28:
Is there a way to close the connection to a conference whenever I close svxlink with killall?
I made a telegram bot to change conference from outside home that changes the autoconnection conference in ModuleEcholink.conf, but when I close svxlink with killall (very rude way, I know) while it was connected, it reconnected on startup. Help!

dl1esk, 2018-12-02:
Tobias already said that this here is no discussion Forum please report only Bugs here !!!
Discussions or Questions here: https://forum.funk-telegramm.de/board/31-svxlink-forum-english/

sm0svx, 2019-07-16:
NOTE: At the moment there are two official channels for discussing SvxLink; the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/svxlink-devel) and the [groups.io discussion forums](https://groups.io/g/svxlink). GitHub issues should not be used for this.

Anyway, the Echolink protocol is constructed in such a way that there is no "connect" protocol message but just a "link active" message. When SvxLink is restarted the other side just keeps on sending "link active" packets until a timeout occurs. If SvxLink is restarted before that timeout expires the incoming "link active" packet will be interpreted as an incoming connection. SvxLink could possibly be modified to send a disconnect packet before it dies but I don't know if it's worth the effort.

A better solution is to use the DTMF_CTRL_PTY to inject DTMF commands. There also is a COMMAND_PTY in the Echolink module but it does not have a CONN command at this moment, only DISC and KILL.