sm0svx/svxlink #426: Release doesn't work, requires .git in a build step

programmin1, 2018-11-02:
After finally setting the build requirements and installing libraries I got an error, which is resolved if you check out from git:

fatal: not a git repository (or any of the parent directories): .git

So the latest release seems to not work from this.

Do you need help building .deb files for this? That would make install much easier.

dl1esk, 2018-12-02:
Tobias already said that this here is no discussion Forum please report only Bugs here !!!
Discussions or Questions here: https://forum.funk-telegramm.de/board/31-svxlink-forum-english/

sm0svx, 2019-07-16:
@programmin1, thanks for pointing that out. It's a bug. However, it does not seem to interfere with the build of the source code other than that there is that ugly printout. "make package" probably will not work but that is experimental anyway. I'll fix this in the next release.

There already is an [official build for Debian and Ubuntu](https://tracker.debian.org/pkg/svxlink). You can [contribute](https://github.com/lechner/svxlink) there if you have ideas for change.

To clarify, the make package thing is not meant to be used to release real packages but rather as a means for those that run SvxLink from source to be able to quickly package SvxLink up for local use.

NOTE: At the moment there are two official channels for discussing SvxLink; the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/svxlink-devel) and the [groups.io discussion forums](https://groups.io/g/svxlink).

sm0svx, 2019-07-16:
Fixed in the master branch

closed 2019-07-16 by sm0svx