sm0svx/svxlink #432: Info Timing Identification

ik4nzd, 2018-12-26:


ik4nzd, 2018-12-26:
To test the audio messages, in SVXLINK.CONF I set these parameters:

SHORT_IDENT_INTERVAL = 1
LONG_IDENT_INTERVAL = 3

Unfortunately there is something that does not work because 2 LONG is sent only 1 SHORT, while I think it should be sent 2.
Use SvxLink v1.5.99.8

See log

SimplexLogic: Sending short identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Sending long identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Sending short identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Sending long identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Sending short identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF
SimplexLogic: Sending long identification ...
Tx1: Turning the transmitter ON
Tx1: Turning the transmitter OFF

Tnx, '73 de Adam

pe1chl, 2018-12-28:
Maybe you hit some boundary case.  Try LONG_IDENT_INTERVAL = 5 and see what happens.

sm0svx, 2019-07-15:
There is a [hard coded limit](https://github.com/sm0svx/svxlink/blob/master/src/svxlink/svxlink/Logic.tcl#L22) that will make SvxLink emit short idents at most with two minutes interval. It's kind of unintuitive but I don't see it as a valid use case or do you really want to have idents every minute? Closing for now.

closed 2019-07-15 by sm0svx