sm0svx/svxlink #433: LocationInfo will not gate RF

la1pha, 2019-01-07:
I'm running SVXLINK as simplex Echolink. I've used EL-"nodenumber" as callsign and it is gated to aprs-is , but not to RF. I think that is because missing passcode.
Is there possibillity to add object owner that should be log in info callsign and passcode entry in LocationInfo section in svxlink.conf?
As a quick fix ! Is there a way to add this directly somewhere where the login string to aprs-is is located?
If I have turned the problem up side down I would like your(anyone) feedback.

dl1hrc, 2019-01-08:
The aprs passcode in SvxLink is generated from the callsign automatically. If the beacon is not sent on Rf the reason maybe the configuration of the aprs gateway. Normally IS traffic is not gated to Rf since it leads to extreme  frequency load.

la1pha, 2019-01-08:
I see. Thanks for fast answer. Then my suggestion is that we have two fields. MYCALL that is owners call and OBJECTNAME. Then passcode will be generated from MYCALL so APRS will show example : EL-nnnnnn (from la1xxx) and be gated to RF. Possibly a switch mycall-nn NoRF to disable passcode. I think every Ham is responible to check how busy aprs is in his area and set beacon time and path accordingly.

dl1hrc, 2019-01-08:
That's a point I never understood in aprs, why we should send the echolink nodenumber over Rf instead of the callsign? I know there is an aprs message definition for "Object from xxx" but in my opinion it should used for passive objects that could not identify by itself. I already had the discussion some of years ago.
Anyway, you can use the SvxLink's aprs functionality to gate your own aprs messages to the network. To do this you should send an aprs string to the device defined as PTY_PATH=/tmp/aprs in the section [LocationInfo], e.g. 
echo "DL1HRC-10>APRS,WIDE1-1:Test from Adi" > /tmp/aprs

la1pha, 2019-01-08:
Hi Adi.
First my appoligies for reporting a bug that clearly was not. SVXLINK is working as it should. I discovered that our Igate wasn't gating MMDVM repeater to RF either. And since I'm the sysop for Igate as well I found a simple solution and get svx status report gated and MMDVM as well. It was a simple matter to include only the objects I wanted in aprx filtersection "m/50 o/LA1PHA-L/LD8MR*" to force the reporting. That solved this problem. Thanks for input and I think twice twice before reporting another bug ;-)

sm0svx, 2019-07-15:
@la1pha, please close issues that have been resolved. Saves me some time. Thanks!

closed 2019-07-15 by sm0svx