sm0svx/svxlink #439: Metar URL has changed

nickbeee, 2019-02-09:
Metar URLs from tgftp.nws.noaa.gov/data/observations/metar/stations/ now use SSL. 
e.g. `https://tgftp.nws.noaa.gov/data/observations/metar/stations/EGNX.TXT`

Change required to src/svxlink-17.12/src/svxlink/modules/metarinfo/ModuleMetarInfo.cpp. ... line 2067.

dl1hrc, 2019-02-10:
to use ssl you need more changes than just that link. I've already built an own branch a couple of month ago, but it's still not in the trunk.
Find further info here: https://github.com/dl1hrc/svxlink/tree/metarssl

73s

nickbeee, 2019-02-11:
Thanks for clarifying that. I did make the change above but all that does is modify the URL written in the log, I can see the request still goes out as http:// to port 80.

dl1hrc, 2019-02-11:
Since you need curl (or an other lib) that's able to handle https connections. It's included in metarssl branch. You may try to use 
ftp://tgftp.nws.noaa.gov/data/observations/metar/stations/
and change http:// to ftp:// in source code. But this is tinkering :/ I don't know how long it will be available on noaa servers. The better way would be to check the branch in into the master trunk someday...

nickbeee, 2019-02-11:
I did try setting up a local proxy with socat to make the connection ,but this failes as the GET request isn't formatted correctly (it sends localhost instead of tgftp.nws.noaa.gov).

`socat tcp-listen:80 OPENSSL:tgftp.nws.noaa.gov:443,verify=0` 

and in ModuleMetarInfo.conf
`TYPE=TXT`
`SERVER=localhost`
`LINK=data/observations/metar/stations`

Although anything achived by this method will be a work-around until your branch gets checked in :)

sp2ong, 2019-03-19:
Hi Tobias

When you can expect a DL1HRC patch in svxlink. The Metar module has stopped working for us. We look forward to hearing from you.

sm3sgp, 2019-03-20:
The fix from Adi/DL1HRC is now in the master branch.
https://github.com/sm0svx/svxlink/blob/e4abe4596635d1ae0b85b46f92ee46599c1f83a6/src/svxlink/ChangeLog

closed 2019-06-24 by sm0svx