sm0svx/svxlink #440: Echolink SvxLink v1.6.99.11, Server v2.5.9997 - Hangs on "transmitter ON"

Rodrigo-NH, 2019-02-20:
Hi. I compiled svxlink from Master in RaspberryPi2.
Using latest Echolink for Android is very recurrent transmitter hanging on “ON” after clicking on end transmission. It happens specially while using a proxy connection, but I saw occurring also while using relay.
Using Echolink on computer things are better but it happened once after some tries (after 10~15 tries)
Unfortunately, I wasn’t able to state this is a server or a client issue but when it happens if I click “END” on android or computer echolink, server releases the transmission promptly.
Going to try older svxlink versions to see what happens.
Thanks

Rodrigo-NH, 2019-02-21:
Testing further and by this time after a hang server kept transmitter “ON” even after disconnecting successfully. I was the only user connected.

 22:11:32 2019: Tx1: Turning the transmitter ON
 22:11:33 2019: Tx1: Turning the transmitter OFF
 22:11:33 2019: Tx1: Turning the transmitter ON
 22:11:41 2019: PU4RNH: EchoLink QSO state changed to BYE_RECEIVED
 22:11:41 2019: PU4RNH: EchoLink QSO state changed to DISCONNECTED
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_p" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_u" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_4" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_r" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_n" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "phonetic_h" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "disconnected" in context "EchoLink"
 22:11:41 2019: SimplexLogic: Deactivating module EchoLink...
 22:11:41 2019: *** WARNING: Could not find audio clip "deactivating" in context "Default"
 22:11:41 2019: *** WARNING: Could not find audio clip "name" in context "EchoLink"

sm0svx, 2019-07-15:
I bet the problem have to do with you not installing the audio clips, hence the warnings. It's possible there is a bug when no audio clips are found.