sm0svx/svxlink #442: Disable rtlsdr receiver when transmitting.

michelvv76, 2019-03-14:
Hello SM0SVX,

We have use your software here in the Netherlands to try to get a network of CB Repeaters by the whole country. We have about 70 RX and TX location that are working toogether with your software. Take a look at our page http://www.cbrepeater.nl. Now i got a question is it possible because we also use RX en TX on one antenne site. With a relay we switch from RX to TX. To make some rules to define that site and disable the RX when TX is transmitting. RX we use RTLSDR dongles. And frequenties that we use are RX: CH11 27.085/ TX: CH18 27.175 so if we build compact repeater setup we not get problems with the RTLSDR catch signals from the TX and not have to make the receiver less sensetive.

Greetings Michel van Veen (19WO1934)

sm3sgp, 2019-03-19:
Are you running a RepeaterLogic or a SimplexLogic? Or is it a remotetrx?
Use a SimplexLogic if you dont want the RX to be active when TX. Also check the parameter "MUTE_RX_ON_TX".

michelvv76, 2019-03-19:
We use Remotetrx for it.

sm3sgp, 2019-03-20:
Sorry, I don't understand the setup.
You have a central RepeaterLogic and many remotetrx. The RepeaterLogic use MULTI_TX to use remote transmitters? This interfering transmitter, is it also connected to the system, but configured in some way so it it not always transmitting? Is it configured in the same remotetrx as the TX?

michelvv76, 2019-03-20:
Sorry for the bad explaination. But we run one svxlink instance on a server in datacenter that take care of al the connection to all repeaters that er raspberry pi's that are running remotetrx to connect to the svxlink server. So we have a central repeaterlogic with many remotetrx. Yes we also use MULTITX for the transmitters. When we transmit on a repeater site with RX an TX on one antenne that is not a great setup but possible, and have make compact repeater setup the RTLSDR is catching signals from the TX so we must make the RTLSDR less sensitive. And that is why i got question can we define that RTX repeaters and then when TX is transmitting on that site disable the RX on that site.

sm3sgp, 2019-03-23:
Thanks for the detailed explanation.
Maybe one solution could be to add one Logic to your SvxLink node, one SimplexLogic. The SimplexLogic has the default behaviour to mute the RX when the TX is active. The RepetearLogic and the Simplex logic can be linked together, (Read about this in the "Logic linking" chapter in the manual).
If you have a remotetrx (with TX and RX) that should act as a simplex link, you can connect it to the SimplexLogic instead of the RepeaterLogic.

sm0svx, 2019-07-15:
The tip from @sm3sgp should solve the problem.

closed 2019-07-15 by sm0svx