sm0svx/svxlink #444: ModuleFrn

dh9yap, 2019-03-25:
Hello ... it's about the ModuleFrn. Would be glad if the voice from the Svxlink (internet) is also transmitted to Freeradio Network Radio. At the moment, only the voice is transmitted from the gateway (local) to the freeradio network. I would be glad about an implementation. 73 Jürgen DH9YAP

dh9yap, 2019-03-27:
@sh123?

dh9yap, 2019-04-21:
Halllloooo

sh123, 2019-04-23:
@dh9yap, can you explain your use case in details, do you want cross link SVXLINK and FRN, so audio streams will flow between them? Is it OK if data from non-amateur radio channels will flow into amateur radio network?

dh9yap, 2019-04-24:
Hello ... Yes we want to connect Frn and svxlink. We use svxlink on Freenet PMR and CB radio ... not in amateur radio .... the audio from Frn is transferred with your ModuleFrn into the svxlink network ... but what from the Svxlink network comes is not heard in FRN ... it would be a very nice option for us ... 73 Jürgen

sh123, 2019-04-25:
@dh9yap, QsoFrn implements Async::AudioSink and Async::AudioSource, which are part of svxlink Async framework, module does have control over audio sink and source, these are provided and configured by root [svxlink.conf](http://www.svxlink.org/doc/man/man5/svxlink.conf.5.html). If you think that problem comes form Frn module, please, share more details, like, your arch setup and svxlink.conf / frn config file.

dh9yap, 2019-04-25:
@ sh123 Hi :-)
Here are our settings.
From other gateways, from the Svlink network, the audio is not transmitted to the Freeradionetwork.
From the freeradion network to the SVxlink gateway, audio is being transmitted.
I drew that on paper. [https://drive.google.com/file/d/1zmHLtzQwCBrXvWuzECW-nuGTSjjZGvZ0/view?usp=sharing](url)

##########################################################

In the old days, it only worked with a little trick.
We had the module integrated in the NetLinkLogic ...
But since we switched to Reflector this does not work.

[NetLogic1]
TYPE=Repeater
RX=NetRx1
TX=NetTx1
MODULES=ModuleFRN
CALLSIGN=NONE
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=de_DE
OPEN_ON_SQL=50
OPEN_SQL_FLANK=OPEN
NO_REPEAT=1
IDLE_TIMEOUT=0
DTMF_MUTING=1 

##################################################################

**and here is the current configuration**
###########################################################################
System Orange PI

ARMBIAN 5.38 stable Debian GNU/Linux 9 (stretch) 4.14.18-sunxi

###########################################################################
[ModuleFrn]
NAME=Frn
PLUGIN_NAME=Frn
ID=7
#TIMEOUT=300

# Details http://freeradionetwork.eu/frnprotocol.htm

# main server
SERVER=funknetzdeutschland.ddns.net
PORT=10024

# backup server
SERVER_BACKUP=europa-welle.de
PORT_BACKUP=10024

# login details
VERSION=2014003
EMAIL_ADDRESS=volker.herrmann65@gmx.de
DYN_PASSWORD=######
CLIENT_TYPE=2
CALLSIGN_AND_USER="13AC1, Volker"
BAND_AND_CHANNEL="446.03125FM CTC131.8"
DESCRIPTION="SvxLink FreeRadioNetwork Station"
COUNTRY=Germany
CITY_CITY_PART="Lengerich - Mitte"
NET=Privat

# debugging
#FRN_DEBUG=1
#DISABLE_RF=1

################################################################################

[GLOBAL]
#MODULE_PATH=/usr/lib/arm-linux-gnueabihf/svxlink
LOGICS=FreenetLogic
CFG_DIR=svxlink.d
TIMESTAMP_FORMAT="%d.%m.%Y %H:%M:%S"
CARD_SAMPLE_RATE=48000
#CARD_CHANNELS=2
#LOCATION_INFO=LocationInfo
LINKS=NetLink1

[FreenetLogic]
TYPE=Simplex
RX=Rx1
TX=Tx1
MODULES=ModuleHelp,ModuleMetarInfo,ModuleFrn
CALLSIGN=45699b
SHORT_IDENT_INTERVAL=10
LONG_IDENT_INTERVAL=60
#IDENT_ONLY_AFTER_TX=4
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=de_DE
#REPORT_CTCSS=67.0
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-19

[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:0
AUDIO_CHANNEL=0
SQL_DET=SIGLEV
SIGLEV_SLOPE=28.46
SIGLEV_OFFSET=-85.27
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
VOX_THRESH=1500
VOX_FILTER_DEPTH=20
SQL_START_DELAY=800
SQL_DELAY=10
SQL_HANGTIME=1000
#SQL_TIMEOUT=600
DEEMPHASIS=0
PREAMP=-7
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=1
DTMF_HANGTIME=40

[Tx1]
TYPE=Local
#AUDIO_DEV=alsa:plughw:0
AUDIO_DEV=alsa:AMP1
AUDIO_CHANNEL=0
PTT_TYPE=SerialPin
PTT_PORT=/dev/ttyUSB0
PTT_PIN=RTS
PTT_HANGTIME=10
TX_DELAY=10
#TIMEOUT=180
PREEMPHASIS=0
DTMF_TONE_LENGTH=100
DTMF_TONE_SPACING=50
DTMF_DIGIT_PWR=-15
MASTER_GAIN=6.0

[ReflectorLogic1]
TYPE=Reflector
HOST=213.202.228.66
PORT=5300
CALLSIGN="Herten-Sued-CB-Kanal-34"
AUTH_KEY="######"
JITTER_BUFFER_DELAY=200

[NetLink1]
NAME=NET
DEFAULT_ACTIVE=1
CONNECT_LOGICS=FreenetLogic:13:GER,ReflectorLogic1:13:GER
TIMOUT=600

######################################################################

Another little info:
in the ModuleFrn.conf the version number has to be changed from VERSION = 2014000 to VERSION = 2014003 otherwise the ModuleFrn will not start.

###############################################################################

sh123, 2019-04-26:
@dh9yap, I think FreenetLogic RX needs to use NetLink otherwise it does not have any chance to get audio stream as currently RX=Rx1 and Rx1 is the audio stream coming from ALSA.

sh123, 2019-04-26:
@dh9yap, current logic looks like this
 
1. incoming: rig -> alsa:plughw:0 -> [Rx1] -> [FreenetLogic] -> [ReflectorLogic1]
2. outgoing: [ReflectorLogic1] -> [FreenetLogic] -> [Tx1] -> alsa:AMP1 -> rig

(1), should work, because Rx1 is the AudioSource for the FRN module, it reads audio from this source and streams it into its network
(2) is not going to work, because Tx1 is the AudioSink, this is where FRN is sending audio coming form its network, it does not read from it, it is not bidirectional

To make it work audio from the ReflectorLogic needs to be routed into RX of the FreenetLogic, so it should be something like RX=NetLink1, but in this case audio from rig won't be routed into FRN.

dh9yap, 2019-04-28:
@sh123 
maybe dl1 HRC can help us I have already contacted him via e-mail. He looks after the project here in DL

dl1hrc, 2019-04-28:
I have no experiences with Frn, just to be sure what you want to realize:
- a local simplex-link on CB or PMR (via Rf)
- a connection to a special CB/PMR-Network via a ReflectorLogic
- connect all of them together
Is this right?

The first thing what strikes me is that ReflectorLogic is not active:
[GLOBAL]
...
LOGICS=FreenetLogic
...
should be 

**LOGICS=FreenetLogic,ReflectorLogic**

But there should be an error message in the svxlink log, isn't it? Please could you give me a short feedback.

73s de Adi / DL1HRC

dl1hrc, 2019-04-28:
The other thing is the alsa definition in your [Tx1] section, it is different from the [Rx1] section and defines a special alsa functionality. What is the reason for this? Did you define an alsa device with name AMP1 in the asound.conf? I recommend to use a "normal" alsa:plughw device first as already used in the [Rx1] section. Before using special alsa extensions the audio stream should flow without any sorcery ;)

dh9yap, 2019-04-28:
Hello Adi
Sorry had modified the svxlink.conf for Github. I forgot to add Reflectorlogic.
Have changed that.
AMP1 is the compressor ...
I have disabled now and everything works with standard.
ssh123 and I suspect that the passage of the audio stream from the Reflector_Netz to the Freeradionetwork is not directly on the module.

How does the Echolink module work?
Is the audio from a Reflectorverbund also not transferred to Echolink?

73 Jürgen

dl1hrc, 2019-04-28:
> Is the audio from a Reflectorverbund also not transferred to Echolink?

No, connections between EchoLink and ReflectorLogic and SimplexLogic are working all the way without any problems. Has it (Frn configured together with a SimplexLogic and ReflectorLogic) ever worked in the past or is it the first try to realize this setup?

dh9yap, 2019-04-28:
@dl1hrc @sh123 
It does not work with the Reflector!
With the SVXlinkserver and a Netlogic it worked only with a trick, described this in a previous text here.
In the old days, it only worked with a little trick.
We had the module integrated in the NetLogic ...
But since we switched to Reflector this does not work.

[NetLogic1]
TYPE=Repeater
RX=NetRx1
TX=NetTx1
MODULES=ModuleFRN
CALLSIGN=NONE
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=de_DE
OPEN_ON_SQL=50
OPEN_SQL_FLANK=OPEN
NO_REPEAT=1
IDLE_TIMEOUT=0
DTMF_MUTING=1

if you compare Echolink and the ModuleFrn you could then see what it is.

dl1hrc, 2019-04-28:
ok, I think we should try to solve the problem via eMail for now. If the problem is solved we will report it here.

dh9yap, 2019-07-14:
Hello to all readers ... Adi is in the process of solving the problem. It will take some time.  At the end of June he wanted to look at the code ... 73

sm0svx, 2019-07-15:
@dl1hrc, do you know what the problem is?  Have you reproduced it locally? SvxLink core or in the module?

dl1hrc, 2019-07-15:
is not really clear at the moment, will see what I can do next days

dl1hrc, 2019-07-17:
yes, it's a bug. The audio stream isn't send over the FRN-connection if received from a connected reflector. 
Local audio stream from Simplex/Repeater-Logic will send to FRN als well as the audio is streamed into the reflector network received from FRN.

dl1hrc, 2019-07-17:
The problem is the State-enum in QsoFrm class or more clear the assigned state depending from the type of audio stream. The local handled audio stream (Simplex/Repeater-Logic) has the state STATE_TX_AUDIO while the audio stream from reflector network gets the state STATE_IDLE, so the writeSamples method ignores the samples received from the reflector network.

dl1hrc, 2019-07-18:
ok, the real problem was to use the local squelch as the only criteria to send the audio stream into the FRN network. If you are using a radio+squelch (vox, ctcss,...) everything works fine, but the audio stream received over the reflector network does not have a squelch criteria, so we have to detect an incoming audio stream independently if it is received local or over the network. In both cases we have to request the permission to send (RQ_TX0) and set the state to STATE_TX_AUDIO_WAITING. The diff is quite easy:

diff --git a/src/svxlink/modules/frn/QsoFrn.cpp b/src/svxlink/modules/frn/QsoFrn.cpp
index 50d5a148..f1168ce8 100644
--- a/src/svxlink/modules/frn/QsoFrn.cpp
+++ b/src/svxlink/modules/frn/QsoFrn.cpp
@@ -373,6 +373,13 @@ std::string QsoFrn::stateToString(State state)
 int QsoFrn::writeSamples(const float *samples, int count)
 {
   //cout << __FUNCTION__ << " " << count << endl;
+
+  if (state == STATE_IDLE)
+  {
+    sendRequest(RQ_TX0);
+    setState(STATE_TX_AUDIO_WAITING);
+  }
+
   int samples_read = 0;
   con_timeout_timer->reset();
 
@@ -438,8 +445,8 @@ void QsoFrn::squelchOpen(bool is_open)
 {
   if (is_open && state == STATE_IDLE)
   {
-    sendRequest(RQ_TX0);
-    setState(STATE_TX_AUDIO_WAITING);
+//    sendRequest(RQ_TX0);
+//    setState(STATE_TX_AUDIO_WAITING);
   }
 }

dh9yap, 2019-07-19:
Very Good  Job !!!

sm0svx, 2019-07-25:
Since @sh123 is the maintainer of the Frn module I'm awaiting a thumbs up on the suggested fix. Thanks for your work Adi!

dh9yap, 2019-09-27:
@sh123
Please respond once to the request of Tobias so that the topic can be closed and the change can be incorporated into the trunk. Its Frn module is much appreciated by many ... it works very well after the cleanup of Adi.

73 dh9yap or 13dkdo03

sh123, 2019-09-27:
yes, do it

sm0svx, 2019-09-27:
Thanks for confirming. Done.

commit 113414b7 (2019-09-27): Frn module one way audio bugfix

Closes #444
closed 2019-09-27 by sm0svx