sm0svx/svxlink #449: make fails, missing curl

Dloranger, 2019-04-18:
just tried to build trunk and got this

-- Found GCrypt: /usr/lib/arm-linux-gnueabihf/libgcrypt.so
-- Found RtlSdr: /usr/lib/arm-linux-gnueabihf/librtlsdr.so
-- Found Tclsh: /usr/bin/tclsh8.6 (found version "8.6")
-- Found TCL: /usr/lib/arm-linux-gnueabihf/libtcl8.6.so
-- Found DL: /usr/lib/arm-linux-gnueabihf/libdl.so
-- Could NOT find CURL (missing:  CURL_INCLUDE_DIR CURL_LIBRARY)
CMake Error at svxlink/svxlink/CMakeLists.txt:48 (message):
  Could not find the CURL library and development files.

-- Configuring incomplete, errors occurred!
See also "/root/svxlink/svxlink/src/build/CMakeFiles/CMakeOutput.log".
See also "/root/svxlink/svxlink/src/build/CMakeFiles/CMakeError.log".
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'doc'.  Stop.
make: *** No rule to make target 'install'.  Stop.
Failed to enable unit: File svxlink.service: No such file or directory
Failed to disable unit: No such file or directory

dl1hrc, 2019-04-18:
> Could not find the CURL library and development files.
libcurl-**dev** is needed

Dloranger, 2019-04-18:
better now, thanks Adi

closed 2019-04-18 by Dloranger