sm0svx/svxlink #450: idents disabled, but still getting idents every couple minutes

Dloranger, 2019-04-19:
working on a setup where the idents are disabled, but during a QSO the system idents every 2 minutes or so when the sql is dropped.

I have chased the issue down to the repeaterLogic.tcl file, and more specifically down to the following chunk of code.

I think this is a bug, so I am posting it here, 

Can anyone explain why this code block 1) gets activated during a qso, and 2) needs to ident at all, 3) why every 2 minutes when the idents are disabled?  

Shouldn't all idents (aka spellWord $myCall) also be handled by triggering the short_id process to honor the setttings the user has selected?  This will also have them logged properly, where these types of IDs are not logged at all. 

proc repeater_down {reason} {
  global mycall;
  variable repeater_is_up;

  set repeater_is_up 0;

  if {$reason == "SQL_FLAP_SUP"} {
    playSilence 500;
    playMsg "Core" "interference";
    playSilence 500;
    return;
  }

  set now [clock seconds];
  if {$now-$Logic::prev_ident < $Logic::min_time_between_ident} {
        return;
  }
  set Logic::prev_ident $now;

  **spellWord $mycall;**
  playMsg "Core" "repeater";
  playSilence 250;

  #playMsg "../extra-sounds" "shutdown";
}

Dloranger, 2019-06-26:
It has been noted that the source of the code is missing, the files that contain this code are the logics such as RepeaterLogic.tcl

To allieviate the problem, we ended up commenting out the entire contents of these two function calls (leave the framework, just make them empty functions) to ensure the system was completely quiet when ids were set to be disabled.

sm0svx, 2019-07-15:
It's really not a bug. The short ident is used for periodical identifications while the function that are causing your problem is a repeater up/down identification. The reason you are getting identifications every two minutes is caused by using a short idle timeout in the repeater configuration so that the repeater often goes up/down in the middle of a QSO.

I know that using a short timeout is common outside Sweden but when using the default longer timeout the repeater normally stays up for a whole QSO so the up/down identifications are not a problem. The repeater at the least identify at the start and at the end of the QSO, and also periodically if configured. Feels quite natural.

Now if the idle timeout is set short it's not so natural anymore. The two minutes you experience in between idents is caused by the variable min_time_between_ident in Logic.tcl. It's just there to stop annoying repeating idents if the repeater goes up/down multiple times in a short while. However, it's not designed to handle configurations using short idle timeouts.

So what's the solution? The TCL files should be seen as advanced configuration files so if something does not behave the way you want you can modify it. Most functions are very short, just playing some sound on some specific event. The identification logic is the exception having comparatively large functions containing a lot of logic. The reason for that design was to enable detailed changes of the identification process. Given that, the "correct" way to solve the problem is to create a **local** subdirectory under the events.d directory where you put a copy of RepeaterLogic.tcl where all  variables and functions but repeater_up and repeater_down has been removed. Now you can have your identification behaviour tailored exactly to your liking.

But I guess your use case is so common that introducing a new configuration variable is warranted, like IDENT_ON_RPT_UP_DOWN, or something like that. If set to 0 nothing is played back on up/down, no ident and no dut-dut.

Dloranger, 2019-07-16:
Thanks Tobias,  

As noted, it would be great if these particular functions could be retooled to point to the user configured  ident functions for the actual announcements so they will behave the same as the other idents.  I tried to figure out how to do this, but I wasn't able to figure it out so thats why we effectively emptied out those functions.  This was the only way we could ensure everything behaved the same.

We are working to upgrade the capabilities of the idents in general such that they will have the options of phonetic or non-phonetic alphabet, time as 12-24 hour, voice and/or CW, etc, so in my mind these idents need to use those functions so the entire ident system behaves the same.

If the user only wants CW (or perhaps is only allowed to have CW for some reason) there shouldn't be some errant function that still forces the voice announcements.  This portion should all be centralized.