sm0svx/svxlink #451: Enable FEC

iw0red, 2019-04-20:
Hi, I found this: 

https://github.com/sm0svx/svxlink/search?q=FEC&type=Code

but I don't know how to enable it,  can anyone show me? 

73 Marco IW0RED.

iw0red, 2019-04-22:
I made this:
first:
vim /root/svxlink/src/async/audio/AsyncAudioEncoderOpus.cpp +131
after that: 
add enableInbandFec(true);
finally: make && make install
Is that right?
73 Marco IW0RED

sm0svx, 2019-07-15:
The Opus codec inband FEC is not settable at this time so modifying the source code is the only way to activate it right now. Your modification should enable the FEC. Did you notice any improvement?

iw0red, 2020-04-24:
Sorry, I lost it.....
I didn't make any misure but it seems work better.
73 Marco IW0RED.