sm0svx/svxlink #454: SVXLink use DVswitch to bridge DMR/P25/NXDN

sp2ong, 2019-05-14:


sp2ong, 2019-05-14:
On Debian 9 we can install DVSwitch in following:
as root user:
 wget http://dvswitch.org/install-dvswitch-repo
chmod +x install-dvswitch-repo
./install-dvswitch-repo
apt update
apt install dvswitch -y

The package dvswitch is installed in /opt/

For configure Analog_Bridge to use md80-emu which deliver transcode for DMR/P25/NXDN/YSF/DStar is edit /opt/Analog_Bridge/Analog_Bridge.ini in [GENERAL]

decoderFallBack = true                  ; Allow software AMBE decoding if a hardware decoder is not found
useEmulator = true                      ; Use the MD380 AMBE emulator for AMBE72 (DMR/YSFN/NXDN)
emulatorAddress = 127.0.0.1:2470        ; IP address and port of the md380 server

This tells Analog_Bridge to connect to the emulator running on the same host on UDP port 2470.
Now questions how to send and receive audio from svxlink to Analog_Bridge.ini:

; Information for USRP channel driver.  This interface uses PCM to transfer audio information
; There are two typical configurations, ASL and Transcode.  ASL (AllstarLink) is for analog clients ;connected
; to a digital network.  Transcode is when Analog_Bridge actually points its PCM interfaces back at ;itself, 
; causing a TLV <-- (pcm <--> pcm) --> TLV type of architecture.
;   When using ASL, this matches the rpt.conf ASL file with a setting like:
;   rxchannel = usrp/127.0.0.1:34001:32001
;   When Transcoding, make toASLPort equal to the other instance fromASLPort (crossover)
[USRP]
server = 127.0.0.1                      ; IP address of Allstar/Asterisk
toASLPort = 32001                       ; Analog_Bridge <-- ASL
fromASLPort = 34001                     ; Analog_Bridge --> ASL
aslAudio = AUDIO_UNITY         ; Audio to ASL (AUDIO_UNITY, AUDIO_USE_AGC,AUDIO_USE_GAIN)
agcGain = -20                           ; Gain (in db) of the AGC filter
dmrAudio = AUDIO_UNITY            ; Audio from ASL (AUDIO_UNITY, AUDIO_USE_GAIN,AUDIO_BPF)
dmrGain = 0.35                          ; Gain factor of audio from ASL (0.0-1.0)

sp2ong, 2019-05-14:
Tobias, I have a question, I found in the RAW_AUDIO_UDP_DEST description in svxlink, is it possible to use it as a stream to transmit and receive audio as it is in the USRP on DV Switch?

sp2ong, 2019-05-15:
I have configured and run DVSwitch on my repeater svxlink. It looks like the connection between DMR BM and DVswitch works on my computer. I found USRPAudio.py in Python, which can send sound from DVSwitch to the sound card https://github.com/Bwtnet/BWTNET1/blob/master/USRPAudio.py, but I do not know how to use, for example, the alsa loop to interact with svxlink. This may be a problem that USRPAudio uses audio 8000 rate

dl1hrc, 2019-05-15:
modeprobe snd-aloop creates the loopbackdevice, you can access it from svxlink:

[TxLoopback]
TYPE=Local
AUDIO_DEV=alsa:plughw:3,1,0
AUDIO_CHANNEL=0

[RxLoopback]
TYPE=Local
ID=1
AUDIO_DEV=alsa:plughw:3,1,1
AUDIO_CHANNEL=0

The other "end" will be alsa:plughw:3,0,0 and alsa:plughw:3,0,1
With alsa you can do resampling, look for asound.confexamples, e.g. https://www.alsa-project.org/wiki/Asoundrc
Try to start the 8k-framework before starting SvxLink, maybe this will work.
vy 55s de Adi / DL1HRC

sp2ong, 2019-05-15:
Adi,

I must select the proper device number for alsa loop in USRPAudio.py use parameter
  For select an other sounddevice as the default, you need to extend the script:
def rxAudioStream():
...
stream = p.open(format=FORMAT,
                    channels = CHANNELS,
                    rate = RATE,
                    output = True,
                    frames_per_buffer = CHUNK,
                    output_device_index=0,
                    )

def txAudioStream():
...
stream = p.open(format=FORMAT,
                    channels = CHANNELS,
                    rate = RATE,
                    input = True,
                    frames_per_buffer = CHUNK,
                    input_device_index=2,
                    )     

List of devices number I can do with the following command:

python -m sounddevice
   0 HDA Intel: AD1882 Analog (hw:0,0), ALSA (0 in, 2 out)
>  1 HDA Intel: AD1882 Alt Analog (hw:0,2), ALSA (2 in, 0 out)
   2 Loopback: PCM (hw:1,0), ALSA (32 in, 32 out)
   3 Loopback: PCM (hw:1,1), ALSA (32 in, 32 out)

sp2ong, 2019-05-28:
After many attempts it looks like USRPAudio.py has serious problems with stability of data transfer to the speaker so it cannot be used for svxlink.  However, the quality of the sound sent by the microphone is very good. It seems that the need to add to the svxlink logic USRP similar to Allstar: https://github.com/AllStarLink/Asterisk/blob/master/asterisk/channels/chan_usrp.c will then allow to smoothly transfer audio from Analog_Bridge to svxlink.

sp2ong, 2019-05-29:
OK I have fixed problem with script USRPAudio.py. It needs to change CHUNK=160 to CHUNK = 320 or more like 640 etc after this I have every transmission in speaker. But implement LOGIC USRP Audio via UDP it maybe will be a better solution. The format frames USRP are:
https://github.com/AllStarLink/Asterisk/blob/master/asterisk/channels/chan_usrp.h

sm0svx, 2019-07-15:
@sp2ong, you probably cannot use the RAW_AUDIO_UDP_DEST configuration variable. It is really only meant to be used for debugging incoming audio. What you could try is to use the "udp" audio device type. From the manual page:

> The  "udp" type is not really an audio device but instead will read and write audio from/to a UDP socket. This can be used to interface SvxLink to all sorts of audio sources/sinks capable of streaming raw audio through UDP. One example usage is to interface SvxLink with GNU Radio.  Example: "udp:127.0.0.1:10000". Note however that the only supported format is raw 16 bit  signed  samples,  two interleved channels. Sampling frequency can be chosen using the CARD_SAMPLE_RATE config variable as usual.

So try to set AUDIO_DEV="udp:127.0.0.1:10000", or something like that. It's not well tested though but could be worth a try.

sp2ong, 2019-07-18:
Tobias thank you for comment. Because SvxLink always works with a sampling rate of 16kHz internally and supported sampling rates are 16000 and 48000 I can not connect svxlink with Ananlog_Bridge which use only 8000 sampling rate. Maybe it is possible to modify USRPAudio.py python to resampling audio but I don't know enough Python to solve this problem. 
I have used MMDVM_Bridge and Ananlog_Bridge / MD380-emulator with DVSwitch Mobile application and works very well and quality audio in DMR is sometimes better than chines DMR radios :-)
The md380-emu (emulator AMBE) works very well.
But generally, if we find a solution  how to get a connection between svxlink and Ananlo_Bridge via USRP will get us gateway from analog to Digital Voice world.

sp2ong, 2019-07-19:
The current version of Ananlog_Bridge ( https://github.com/DVSwitch/Analog_Bridge ) can use on UDP port to receive and send stream audio via USP protocol

[USRP]
address = 169.254.42.42                
txPort = 32001                          
rxPort = 32001

We can use IP address 169.254.42.42  ( APIPA ) which allow us simple use firewall to redirect stream analog from and to Ananlog_Baridge across a network. For this reason application, for example, svxlink can send analog audio via USRP across network to Ananlog_Birdge if we have instaled MMDVM_Brideg and Annalog_Bridge on other computer

sp2ong, 2019-11-14:
With the new version of svxreflector which has a talk group, it would be an interesting solution to run svxlink which would have only an audio link via USRP to Ananlog_Bridge DMR gateway with own talk group number then you could choose from other nodes this talk group to have analog <-> DMR connections.

sp2ong, 2019-12-21:
Hi,
I rewrote USRPAudio.py to use with svxlink (compiled with internal sample rates audio 8 kHz) and monitoring PTT and SQL type PTY by script, audio via ALSA Loop. By the last 2 days all working very well and audio is quite good ( I use only md380-emu but maybe when switch to DV3000 it will be better). 
It looks we can use Analog_Bridge/MMDVM_Bridge tools to bridge svxlink with DV world.

sp2ong, 2020-01-06:
I installed allstarlink with echolink module which use the same tools MMDVM_Bridge/Ananlog_Bridge/md380-emu and compare audio quality it is better than audio from svxlink to DMR. I think reason is audio in allstarlink is incoming via codec GSM from echolink application and next to converted to analog and send to Analog_Bridge

sp2ong, 2020-02-03:
I added function resample from 48 kHz to 8 kHz in my svxbridge.py and now I can use typical svxlink with 48 kHz internal sample rate to send and receive audio via ASLA loop and svxbridge.py <-> Analog_Bridge tools

f5vmr, 2020-07-26:
I would like to see perhaps the MMDVM_Bridge/Analog_Bridge talking as a NetRx/NetTx into an existing repeater installation to begin with, restricted to a single TalkGroup - This uses UDP port 5210 if I remember correctly. I've installed the Bridge software on a stand_alone server for the moment, I don't think for one moment that a Raspberry Pi of any model would be capable of driving both the Repeater on SVXLink and the Bridge software at the same time. What are your comments on a possible configuration in either circumstance. The current repeater I'm considering is a Pi 2B, the Pi 4B gets far too hot for my liking.

f5nlg, 2020-07-26:
it works perfectly on orangepi 0 see spotnik v4 http://f5nlg.wordpress.com
, a raspberry image is in progress .

73" de Jean-philippe F5NLG

ur3qjw, 2021-01-28:
Hi,
A very interesting topic.
Is it possible to see the current code somewhere ?
And also a description of the correct setting ? 

Regards, Volodymyr

iu0bvp, 2021-02-22:
Hi .73 I installed dvswitch and svxlink they both work, but I can't get the audio to pass between Dvswitch and svxlink where is the error? What is the right setting? Thank you

sp2ong, 2021-04-16:
I think that the forum will be a better place to discuss this topic: 

[https://groups.io/g/svxlink/topic/75607288](https://groups.io/g/svxlink/topic/75607288)

closed 2021-04-16 by sp2ong