sm0svx/svxlink #455: svxlink after update crash

sp2ong, 2019-05-17:
I have update sources to latest version svxlink and after startup and make echolink connections svxlink broken with message on screen

Incoming EchoLink connection from SP2ONG (Waldek) at 217.97.47.139
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Przerwane

dh9yap, 2019-05-19:
Hello
I have the same problem !
I can activate the ModulEchoLink only if I enter the Node number and confirm with # there is a most complete Svxlink abort.

dj9ph, 2019-05-19:
Hi together,

same problem at DM0JL after todays update:

DTMF command received in module EchoLink: 31
Creating random connection to node:
KB1MMR-R       ON   14:53 Durham, CT                    313834 73.159.240.103
Connecting to KB1MMR-R (313834)
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

I have exactly the same issue at DB0SZ with also running an update today.

Best regards, Christian (DJ9PH)

iw0red, 2019-06-07:
Hi, any news on this?
73 Marco IW0RED

dl1hrc, 2019-06-07:
I've sent a message to the devel-list some days ago:

Hi Tobias,
the reason is in svxlink/modules/echolink/QsoImpl.cpp, line 223:
event_handler = new EventHandler(event_handler_script, 0);
To create an event handler instance the second param (here 0) must be a string (logic_name) since the constructor is defined as 
EventHandler(const std::string& event_script, const std::string& logic_name);

Maybe
event_handler = new EventHandler(event_handler_script, module->logicName());
solves it.

vy 73s de Adi / DL1HRC

dj9ph, 2019-06-08:
Hi together,

just tried the fix - and it worked! thanks Adi for the help!

iw0red, 2019-06-08:
Fine, it works.
Thanks.
Marco IW0RED.

dh9yap, 2019-06-11:
Thank you Adi

sm0svx, 2019-06-24:
Thanks Adi! Your suggested fix saved me a lot of time. The bug should now be fixed in the master branch.

closed 2019-06-24 by sm0svx