sm0svx/svxlink #457: 'std::logic_error'   what():  basic_string::_M_construct null not valid Abortado

LU2ICA, 2019-06-24:
After the last version of Raspian and SvxLink I have the following problem when a station wants to connect to me:
Incoming EchoLink connection from LU2ICA-L ([listen) at 131.108.142.78
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Abortado

sm3sgp, 2019-06-24:
How did you install svxlink? By package with apt install?
What instruction did you follow?

f5vmr, 2019-06-24:
This appears very similar to the issue discussed in the mailing lists with a solution supplied by DL1HRC. There is a problem in the update of may 10, and the solution is listed in this listing

LU2ICA, 2019-06-24:
Good afternoon.
I installed it in the following way: (I do not know which is the correct way)
sudo apt-get update && sudo apt-get upgrade

sudo apt-get install g++ make libsigc++-1.2-dev libgsm1-dev libpopt-dev tcl8.5-dev libgcrypt-dev libspeex-dev libasound2-dev alsa-utils libqt4-dev

sudo apt-get install libsigc++

sudo apt-get install cmake

sudo apt-get install groff

sudo apt-cache search curl

sudo apt-get install libcurl4-openssl-dev

sudo adduser svxlink

sudo wget https://github.com/sm0svx/svxlink/archive/master.tar.gz

tar xvzf master.tar.gz

cd svxlink-master/src

mkdir build

cd build

sudo cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc \
      -DLOCAL_STATE_DIR=/var -DUSE_QT=NO ..

sudo make

sudo make doc

sudo make install

sudo ldconfig

wget https://github.com/sm0svx/svxlink-sounds-en_US-heather/releases/download/14.08/svxlink-sounds-en_US-heather-16k-13.12.tar.bz2

sudo tar -C /usr/share/svxlink/sounds -xvjf svxlink-sounds-en_US-heather-16k-13.12.tar.bz2

sudo mv /usr/share/svxlink/sounds/en_US-heather-16k /usr/share/svxlink/sounds/en_US

First I had a problem when I got to the line:
sudo cmake -DCMAKE_INSTALL_PREFIX=/usr -DSYSCONF_INSTALL_DIR=/etc \
      -DLOCAL_STATE_DIR=/var -DUSE_QT=NO ..
They told me to update the following:
sudo apt-cache search curl

sudo apt-get install libcurl4-openssl-dev
It worked and then not.
Now I would like to know what I should do correctly once I install the latest raspian image that was updated on June 20, 2019. ???

LU2ICA, 2019-06-24:
Incoming EchoLink connection from LU2ICA-L ([listen) at 131.108.142.78
Spurious audio packet received from 131.108.142.78
Incoming EchoLink connection from LU2ICA-L ([listen) at 131.108.142.78
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Abortado
pi@raspberrypi:~ $

sm0svx, 2019-06-24:
Probably the same issue as #455 which is now resolved in the master branch.

LU2ICA, 2019-06-24:
I made that change on line 223 but it does not work.
I would like you to tell me what is the correct way to install Svxlink in raspian and start from scratch.

sm0svx, 2019-06-24:
The only Raspbian specific instruction I know of is the one describing installation using packages: https://github.com/sm0svx/svxlink/wiki/InstallBinRaspbian. But you won't get the latest development code then.

sm3sgp, 2019-06-24:
Please note, the current instruction is NOT verified for the latest release of Raspbian.

Dloranger, 2019-06-24:
Marcello

You can use the open repeater script to install from trunk if desired, this
is used frequently by the OpenRepeater developers so we maintain it with
other misc patches as well to known open issues with svxlink that we can
fix from the command line

https://github.com/OpenRepeater/scripts

Just follow the readme

Dan, kg7par

sm0svx, 2019-06-24:
Try to clone the source tree from GitHub from scratch and recompile, if nothing else helps.

LU2ICA, 2019-06-24:
> The only Raspbian specific instruction I know of is the one describing installation using packages: https://github.com/sm0svx/svxlink/wiki/InstallBinRaspbian. But you won't get the latest development code then.

When I follow these instructions, it gives me this:
pi@raspberrypi:~ $ echo 'deb http://mirrordirector.raspbian.org/raspbian/ buster main' | sudo tee /etc/apt/sources.list.d/svxlink.list
deb http://mirrordirector.raspbian.org/raspbian/ buster main
pi@raspberrypi:~ $

LU2ICA, 2019-06-24:
> Try to clone the source tree from GitHub from scratch and recompile, if nothing else helps.

I'll try..

LU2ICA, 2019-06-24:
I recently reinstalled from the F5VMR & G4NAB page, and it worked!
Thank you very much to all!

closed 2019-06-24 by sm0svx