sm0svx/svxlink #459: Requesting an updated released version

Dloranger, 2019-07-03:
As part of the Open Repeater project team, we would like to request an official release version update.

We only use the officially released "stable" version so our builds are consistent.  

We have been using the current version since early 2018, but a lot has been changed from this last version that we would like to bring into our builds. 

Much appreciated if we could get a fresh locked down revision.

sm0svx, 2019-07-15:
Yes it's been too long (again). My intention is to get a release out soon.

closed 2019-07-15 by sm0svx

Dloranger, 2020-12-07:
@sm0svx Can we reopen this until we get an updated release? Going on 15 months since the last release

Dloranger, 2022-04-15:
Reopening to get updated code into a released version