sm0svx/svxlink #461: raspberrypi svxlink: Libgcrypt warning: missing initialization - please fix the application

f5swb, 2019-07-22:
hello, I'm running buster with the last svxlink git. 
All is ok but I have this in the svxling.log /var/log ?
raspberrypi svxlink: Libgcrypt warning: missing initialization - please fix the application
root@raspberrypi:~# sudo apt install libgcrypt20-dev
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances
Lecture des informations d'état... Fait
libgcrypt20-dev est déjà la version la plus récente (1.8.4-5). "already the newest version"
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.

any idea ?

73's F5SWB.

dh9yap, 2019-07-22:
Hello ... do not understand the error ... is the error message in Svxlink.log to see?  and how does the error affect the operation of svxlink?  ... 73 Jürgen

f5swb, 2019-07-22:
hello Jürgen , many thanks for rou answer
it does not affect the operation of svwlink, nothing in the svxlink.log about that 
I just found this in /var/log/user.log .... so I think that I shouldn't worry of it :)

closed 2019-07-29 by f5swb