sm0svx/svxlink #462: Roger Beep when SQL close on ModuleEchoLink

mathieubourassa, 2019-07-24:
SvxLink release 17.12.2
There is always a Roger Beep at SQL closing when I am listening with a EchoLink client.
The variable RGR_SOUND_DELAY is set on -1 or comment (#) but its make no difference
I have try with the RX1 in VOX or GPIO mode but its make no difference
I am not shure if its normal or a simple issue

There is my configuration in brief;

[SimplexLogic]
TYPE=Simplex
RX=Rx1
TX=Tx1
MODULES=ModuleEchoLink
CALLSIGN=EL-VA2MBP
SHORT_IDENT_INTERVAL=0
LONG_IDENT_INTERVAL=0
#IDENT_ONLY_AFTER_TX=4
#EXEC_CMD_ON_SQL_CLOSE=500
EVENT_HANDLER=/usr/share/svxlink/events.tcl
DEFAULT_LANG=fr_CA
RGR_SOUND_DELAY=-1
#REPORT_CTCSS=136.5
#TX_CTCSS=ALWAYS
MACROS=Macros
FX_GAIN_NORMAL=0
FX_GAIN_LOW=-12
#ACTIVATE_MODULE_ON_LONG_CMD=4:EchoLink
#QSO_RECORDER=8:QsoRecorder
#ONLINE_CMD=998877
#MUTE_RX_ON_TX=1
#MUTE_TX_ON_RX=1
#STATE_PTY=/var/run/svxlink/state
#DTMF_CTRL_PTY=/dev/shm/simplex_dtmf_ctrl

[Rx1]
TYPE=Local
AUDIO_DEV=alsa:plughw:1
AUDIO_CHANNEL=0
SQL_DET=VOX
SQL_START_DELAY=0
SQL_DELAY=0
SQL_HANGTIME=3000
#SQL_EXTENDED_HANGTIME=1000
#SQL_EXTENDED_HANGTIME_THRESH=15
#SQL_TIMEOUT=600
VOX_FILTER_DEPTH=20
VOX_THRESH=1000
#CTCSS_MODE=2
#CTCSS_FQ=136.5
#CTCSS_SNR_OFFSET=0
#CTCSS_OPEN_THRESH=15
#CTCSS_CLOSE_THRESH=9
#CTCSS_BPF_LOW=60
#CTCSS_BPF_HIGH=270
SERIAL_PORT=/dev/ttyS0
SERIAL_PIN=CTS
#SERIAL_SET_PINS=DTR!RTS
#EVDEV_DEVNAME=/dev/input/by-id/usb-SYNIC_SYNIC_Wireless_Audio-event-if03
#EVDEV_OPEN=1,163,1
#EVDEV_CLOSE=1,163,0
#GPIO_PATH=/sys/class/gpio
#GPIO_SQL_PIN=gpio30
#PTY_PATH=/tmp/rx1_sql
#HID_DEVICE=/dev/hidraw3
#HID_SQL_PIN=VOL_UP
#SIGLEV_DET=TONE
SIGLEV_SLOPE=1
SIGLEV_OFFSET=0
#SIGLEV_BOGUS_THRESH=120
#TONE_SIGLEV_MAP=100,84,60,50,37,32,28,23,19,8
SIGLEV_OPEN_THRESH=30
SIGLEV_CLOSE_THRESH=10
DEEMPHASIS=0
#SQL_TAIL_ELIM=300
#PREAMP=6
PEAK_METER=1
DTMF_DEC_TYPE=INTERNAL
DTMF_MUTING=0
DTMF_HANGTIME=40
DTMF_SERIAL=/dev/ttyS0
#DTMF_PTY=/tmp/rx1_dtmf
#DTMF_MAX_FWD_TWIST=8
#DTMF_MAX_REV_TWIST=4
#1750_MUTING=1
#SEL5_DEC_TYPE=INTERNAL
#SEL5_TYPE=ZVEI1
#FQ=433475000
#MODULATION=FM
#WBRX=WbRx1

[ModuleEchoLink]
NAME=EchoLink
ID=2
TIMEOUT=60
#ALLOW_IP=192.168.1.0/24
#DROP_INCOMING=^()$
#REJECT_INCOMING=^()$
ACCEPT_INCOMING=^(VA2MBP)$
#REJECT_OUTGOING=^()$
#ACCEPT_OUTGOING=^(.*)$
#REJECT_CONF=0
#CHECK_NR_CONNECTS=2,300,120
SERVERS=naeast.echolink.org nasouth.echolink.org servers.echolink.org backup.echolink.org servers.echolink.org
CALLSIGN=VA2MBP-L
PASSWORD=MyPassword
SYSOPNAME=VA2MBP-L
LOCATION=[Svx] 146,790MHz La Tuque
#PROXY_SERVER=the.proxy.server
#PROXY_PORT=8100
#PROXY_PASSWORD=PUBLIC
#BIND_ADDR=10.20.30.40
MAX_QSOS=10
MAX_CONNECTIONS=11
#LINK_IDLE_TIMEOUT=0
#AUTOCON_ECHOLINK_ID=9999
#AUTOCON_TIME=1200
#USE_GSM_ONLY=1
#DEFAULT_LANG=en_US
#COMMAND_PTY=/dev/shm/echolink_ctrl

Thanks for your great work

VA2 MBP

sm0svx, 2019-07-25:
Please do not use this issue tracker to ask questions since it's used for bug reports and feature requests only. Use the [SourceForge mailing lists](https://sourceforge.net/p/svxlink/mailman/) or the [groups.io discussion forum](https://groups.io/g/svxlink) for that.

You are probably looking for the REMOTE_RGR_SOUND=0 configuration variable available in ModuleEchoLink. Please have a look at the ModuleEchoLink manual page for more information.

closed 2019-07-25 by sm0svx

mathieubourassa, 2019-08-03:
I am still getting some Roger Beep in both directions with the variable REMOTE_RGR_SOUND=0 in the ModuleEcholing.conf and the RGR_SOUND_DELAY=-1 in the svxlink.conf