sm0svx/svxlink #465: keep old method AUTH_KEY in ReflectorLogic

f5nlg, 2019-09-10:
hello we run about 100 link in our network , password for individual link is not a good idea ...
whe use firewall for remove / accept them 

please have a look at our dashboard of RRF  http://rrf.f5nlg.ovh

so a patch will ben nice for those want keep the old AUTH_KEY autentification with new version 

TK4LS made one ( ReflectorClient.cpp ) if you want it ...

73 de F5NLG

sm3sgp, 2019-09-10:
It is still possible to use the same password for all users. 
From the manual:

   USERS and PASSWORDS sections
       All  users  must  be specified in the USERS section. Usernames are totally arbitrary but typically the callsign is used. The purpose is to map each username to a password. However, the password is not given directly but instead a  reference  to  the PASSWORDS section is given. It's done like this so that multiple nodes can share a single password.

       An example of how the USERS and PASSWORDS sections may look is given below.
         [USERS]
         SM0ABC-1=MyNodes
         SM0ABC-2=MyNodes
         SM1XYZ=SM1XYZ

         [PASSWORDS]
         MyNodes="A very strong password!"
         SM1XYZ="Another very good password?"

       This  will  set up SM0ABC-1 and SM0ABC-2 to use the same password specified by MyNodes in the PASSWORDS section. User SM1XYZ have his own password.

f5nlg, 2019-09-11:
yes thanks i know that , but when you have 100 Link + i let you the pleasure to keep a list of 100+ link who sometimes change whe put (35) CALLSIGN V in CALLSIGN (35) is the departement , callsign + frequency band 10M V U H etc ...

sp2ong, 2019-09-11:
@F5NLG  nice dashboard. Is it an opensource solution? Is it possible to get sources dashboard to install on my local small network?

73 Waldek SP2ONG

f5nlg, 2019-09-11:
Hello Waldek and all , 
yes the dashboard is a part of https://github.com/spotnik-ham who is a set of scripts ,who offer "rooms" in a svxreflecor environement,  dashboard, "hotspot" with OrangePi 0 , shematics and hardware for raspberry and various mods for transceivers . around EchoLink and SvxReflector , our RRF network is only via radio at both end and the heart of system is svxlink , this network exist from about 5 years , and now there is "clones" in Holland , Italy , Portugal , an International room exist , others is localised by Language / Country .
the master page of spotnik distrib is here ( in french ) https://f5nlg.wordpress.com/category/hotspot/
Thanks to SM0SVX and other for this nice software !
we run about 4 svxlink and 4 svxreflector in each server ( 3 server for now ) 

for the complet dashboard with vu meter etc ... you have to ask Jean-François , F1EVM he is in charge of this part if you want ask him directly  he is in copy at this mail  [ admin AT f5nlg DOT ovh ]
[image]

73" de F5NLG / Jean-Philippe.