sm0svx/svxlink #470: svxlink reflector_tg version warning during compile

sp2ong, 2019-10-16:
Hi Tobias,

During complie svxlink reflector_tg version on Raspbian v10 with update sources 15 October 
I have see warning:

[ 84%] Building CXX object svxlink/svxlink/CMakeFiles/svxlink.dir/DtmfDigitHandler.cpp.o
[ 84%] Building CXX object svxlink/svxlink/CMakeFiles/svxlink.dir/ReflectorLogic.cpp.o
In file included from /opt/src/svxlink/src/svxlink/svxlink/ReflectorLogic.cpp:60:
/opt/src/svxlink/src/svxlink/svxlink/../reflector/ReflectorMsg.h: In member function ‘virtual void ReflectorLogic::remoteReceivedPublishStateEvent(LogicBase*, const string&, const string&)’:
/opt/src/svxlink/src/svxlink/svxlink/../reflector/ReflectorMsg.h:994:22: warning: ‘*((void*)& rx +6)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           m_flags = (m_flags & ~bit) | (is_enabled ? bit : 0);
                      ^~~~~~~
/opt/src/svxlink/src/svxlink/svxlink/../reflector/ReflectorMsg.h:994:22: warning: ‘*((void*)& rx +6)’ may be used uninitialized in this function [-Wmaybe-uninitialized]
           m_flags = (m_flags & ~bit) | (is_enabled ? bit : 0);
                      ^~~~~~~
[ 85%] Linking CXX executable ../../bin/svxlink

sm3sgp, 2019-11-03:
The reflector_tg is now merged to master.
This should be fixed.

sp2ong, 2019-11-04:
yes, thank you

closed 2019-11-04 by sp2ong