sm0svx/svxlink #471: Svxlink reflector_tg Compile Error

dj1jay, 2019-10-20:
Hi Tobias,

what am I doing wrong?? Rasbian Buster and Debian same problem!!

[ 54%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/NetRx.cpp.o
[ 54%] Building CXX object svxlink/trx/CMakeFiles/trx.dir/Voter.cpp.o
/root/svxlink/src/svxlink/trx/Voter.cpp: In member function ‘void Voter::printSquelchState()’:
/root/svxlink/src/svxlink/trx/Voter.cpp:713:3: error: ‘StreamWriterBuilder’ is not a member of ‘Json’
   Json::StreamWriterBuilder builder;
   ^
/root/svxlink/src/svxlink/trx/Voter.cpp:714:3: error: ‘builder’ was not declared in this scope
   builder["commentStyle"] = "None";
   ^
/root/svxlink/src/svxlink/trx/Voter.cpp:716:3: error: ‘StreamWriter’ is not a member of ‘Json’
   Json::StreamWriter* writer = builder.newStreamWriter();
   ^
/root/svxlink/src/svxlink/trx/Voter.cpp:716:23: error: ‘writer’ was not declared in this scope
   Json::StreamWriter* writer = builder.newStreamWriter();
                       ^
/root/svxlink/src/svxlink/trx/Voter.cpp:719:10: error: type ‘<type error>’ argument given to ‘delete’, expected pointer
   delete writer;
          ^
svxlink/trx/CMakeFiles/trx.dir/build.make:215: recipe for target 'svxlink/trx/CMakeFiles/trx.dir/Voter.cpp.o' failed
make[2]: *** [svxlink/trx/CMakeFiles/trx.dir/Voter.cpp.o] Error 1
CMakeFiles/Makefile2:2211: recipe for target 'svxlink/trx/CMakeFiles/trx.dir/all' failed
make[1]: *** [svxlink/trx/CMakeFiles/trx.dir/all] Error 2
Makefile:137: recipe for target 'all' failed
make: *** [all] Error 2

dh9yap, 2019-10-31:
@sm0svx Hi
Have the same problem.
Only on an Ubuntu Server LTS 18 it can be installed

sm3sgp, 2019-11-03:
The reflector_tg branch is now merged to master.

Tobias commented on this on the groups.io mailinglist:
_There is one new build dependency added which is libjsoncpp. For deb based systems install with apt install libjsoncpp-dev. Older versions of Debian may not have a sufficiently recent package so you need to upgrade the OS in that case or compile libjsoncpp from source code. The oldest version I have tested with is 1.6.5._

closed 2019-11-30 by sm0svx