sm0svx/svxlink #473: New svxlink on raspbian Buster

Ederion70, 2019-12-25:
Hi,
I'm trying to compile and install the new svxlink 19.09.1 on a raspberryPI 4 with raspbian Buster.
When I use the command "sudo cmake -DUSE_QT=OFF .." in the build directory, at the end I receive this error: "Could NOT find CURL (missing: CURL_INCLUDE_DIR CURL_LIBRARY) 
CMake Error at svxlink/svxlink/CMakeLists.txt:48 (message):
  Could not find the CURL library and development files."
How can I solve it ?
Thank you all!

f5vmr, 2019-12-25:
> On 25

There are some additional packages to install prior to compilation.
Add this line to your rebuild and everything will work.
sudo apt install curl libcurl4-openssl-dev libjsoncpp-dev

Seasons greetings.

Chris

Ederion70, 2019-12-25:
Thank you very much!
I've compiled svxlink perfectly!
I have a question: Can I use the sound clips at this address "https://github.com/sm0svx/svxlink­sounds­en_US­heather/releases/download/18.03.1/svxlink­sounds­en_US­heather­16k­18.03.1.tar.bz2" or I need another version?
Thanks!

Ederion70, 2019-12-25:
Hi,
I found this newer version "https://github.com/sm0svx/svxlink-sounds-en_US-heather/releases/download/19.09/svxlink-sounds-en_US-heather-16k-19.09.tar.bz2" and I used it.
It's playing ok, so I think it's ok.
Now the last question: Where can I find sound clips in Italian language? If they do not exist, where can I find a guide to change the clips and re-recording in italian language?
Thank you very much and Merry Christmas!
Fabio

dl1hrc, 2019-12-25:
https://github.com/dl1hrc/svxlink-sounds-it_IT-chiara

vy 73s es merry xmas de Adi / DL1HRC

Ederion70, 2019-12-28:
Hi Adi,
I beg your pardon. I downloaded sounds chiara and I unzipped all directories in one named it_IT, but it doesn't play and I have a lot of errors like: "*** WARNING: Illegal WAV file header in "/usr/local/share/svxlink/sounds/it_IT/Default/activating.wav". "fmt" subchunk size is not 16: /usr/local/share/svxlink/sounds/it_IT/Default/activating.wav".
How can I solve it?
Thank ypu very very much!!

closed 2020-01-01 by Ederion70