sm0svx/svxlink #477: undefined symbol: _ZTI6Module on RPI2B

oe5hpm, 2020-01-16:
Built with buildroot-environment for rpi2

$ svxlink
SvxLink v1.7.99.20 Copyright (C) 2003-2019 Tobias Blomberg / SM0SVX

SvxLink comes with ABSOLUTELY NO WARRANTY. This is free software, and you are
welcome to redistribute it in accordance with the terms and conditions in the
GNU GPL (General Public License) version 2 or later.

Using configuration file: /etc/svxlink/svxlink.conf
--- Using sample rate 16000Hz

Starting logic: RepeaterLogic
Loading RX: Rx1
Loading TX: Tx1
Loading module "ModuleHelp" into logic "RepeaterLogic"
*** ERROR: Failed to load module ModuleHelp into logic RepeaterLogic: /usr/lib/svxlink/ModuleHelp.so: undefined symbol: _ZTI6Module
Loading module "ModuleParrot" into logic "RepeaterLogic"
*** ERROR: Failed to load module ModuleParrot into logic RepeaterLogic: /usr/lib/svxlink/ModuleParrot.so: undefined symbol: _ZTI6Module
Loading module "ModuleEchoLink" into logic "RepeaterLogic"
*** ERROR: Failed to load module ModuleEchoLink into logic RepeaterLogic: /usr/lib/svxlink/ModuleEchoLink.so: undefined symbol: _ZTI6Module
Loading module "ModuleTclVoiceMail" into logic "RepeaterLogic"
*** ERROR: Failed to load module ModuleTclVoiceMail into logic RepeaterLogic: /usr/lib/svxlink/ModuleTcl.so: undefined symbol: _ZTI6Module
RepeaterLogic: Event handler script successfully loaded.

sm0svx, 2020-01-19:
Sorry, I have no idea. I have not tried to build SvxLink on buildroot for a long time.

oe5hpm, 2020-01-19:
Ok. I will investigate

oe5hpm, 2020-01-29:
I've tracked this down to a single commit within buildroot, they bumped cmake version 3.9.6 to 3.15.4.
Atfer this the "_ZTI6Module" cannot be found anymore.

I will try to isolate this more, maybe we have to change something to be compliant with cmake 3.15.4

sm3sgp, 2025-03-17:
@oe5hpm any update on this?