sm0svx/svxlink #479: Cmake under Debian10 Buster fails

hb9pae, 2020-02-17:
Hi all

Cmake fails under Debian 10, Buster
-- Could NOT find CURL (missing: CURL_INCLUDE_DIR CURL_LIBRARY) 
CMake Error at svxlink/svxlink/CMakeLists.txt:48 (message):
  Could not find the CURL library and development files.

It seems that SVXLINK Source require CURL3, on Debian 10 CURL4 is installed instead.

Any suggestion?

73 de Peter HB9PAE

lechner, 2020-02-17:
Hi Peter,

I maintain the Debian package. Did you declare the prerequisite `libcurl4-gnutls-dev`?

You may be able to use my packaging from `testing`. Please try:
```
dget https://deb.debian.org/debian/pool/main/s/svxlink/svxlink_19.09.1-2.dsc
cd svxlink-19.09.1
debuild -b -uc
```
As an alternative, I could offer a backport of 19.01.1 to `buster`, if you need it.

Kind regards
Felix Lechner WU8K

hb9pae, 2020-02-21:
Thanks Felix
At the moment I can't verify if libcurl4-gnutls-dev was installed, I'll check that on occasion. 
In the meantime, I will close the incident.

73 en Peter HB9PAE

closed 2020-02-21 by hb9pae