sm0svx/svxlink #481: ModuleEchoLink - ACCEPT_INCOMING / REJECT NOT WORKING

jcrix, 2020-03-12:
v1.7.99.21

What I'm trying to accomplish is accept connections only from those callsigns listed in the ACCEPT_INCOMING whitelist.  It does not appear to be working....

Rejecting connection from (CALLSIGN) permanently

Yet client is still able to transmit through from echolink???

ACCEPT_INCOMING=(LIST OF ACCEPTED CALL SIGNS HERE SEPERATED BY PIPES)
REJECT_OUTGOING=^()$
#ACCEPT_OUTGOING=
REJECT_CONF=1

What am I missing?

jcrix, 2020-03-13:
Anybody?

sq4bja, 2020-03-26:
On my svxlink it doesn`t work too. Everyone can connect.

sq4bja, 2020-11-26:
There is an error in some sources. Let`s look into logs:
```
Spurious audio packet received from 172.242.242.42
Incoming EchoLink connection from SQ4BJA (samsung) at 172.242.242.42
Rejecting connection from SQ4BJA permanently
SQ4BJA: EchoLink QSO state changed to CONNECTED
--- EchoLink info message received from SQ4BJA ---

```
There is a reject permanently and accept on another line. Something is wrong.
The same is in the newest version of FM-LINK. I try to find changes in sources from 2017 yaer version, but it looks the same in echolink modules.

Tom SQ4BJA

sm0svx, 2020-11-27:
This may partly be more of a documentation bug rather than a code bug and a bit of misunderstanding. The difference between REJECT and DROP is that the latter will just ignore the connect request while the the former will first accept the connection in order to play a message and when the message has been played the connection will be closed. ACCEPT work similar to REJECT only in reverse.

A rejected connection will be logged as "Rejecting connection from..." then it will go to state CONNECTED in order to play the message and then to DISCONNECTED when the message has been played. This may not work if you have messed with your TCL scripts or sound files so that the message is not being played. But I also have a feeling that there may be a bug in this logic. It should not fail every time though. If it does it's probably a configuration problem.

So if you just want to accept connections from a list of stations, leave DROP and REJECT commented out and just set ACCEPT. The documentation bug I mentioned above, still unverified though, I believe is that all callsigns must be enclosed in parenthesis for the match expression to work. Also, you need to start all regular expressions with a ^ to match from the beginning and a $ to match up until the end of the callsign. Otherwise you may get partial matches that is accepted.

So try something like this:
#DROP_INCOMING=
#REJECT_INCOMING=
ACCEPT_INCOMING=^(CALLSIGN1)|(CALLSIGN2)|(LINK-L)$

With this new information, try to adjust your configuration accordingly. If it works, please close the issue. If it doesn't work I'll have a look at it more closely.

sq4bja, 2020-11-27:
Thank you very much for the advice. The problem was not the ACCESS_INCOMING field syntax, but the corrections in EchoLink.tcl. I tried to turn off the voice prompts. Unfortunately, hashing lines from playMsg caused an error in the "reject_remote_connection" procedure. Everything is working fine now. So I close the issue. However, a handy feature would be to disable all voice prompts with one line of configuration.

73! Tom SQ4BJA

sm0svx, 2020-11-27:
Let's see if @jcrix have any comments before closing.

jcrix, 2020-11-28:
Well i certainly did not have the regular expression written as suggested. I think thats was probably the issue.

sm0svx, 2020-11-28:
ok I'll close it then. Reopen it or create a new issue if you still think there is a bug.

closed 2020-11-28 by sm0svx