sm0svx/svxlink #484: MetarInfo localization ready

circool, 2020-04-02:
Hello Tobias and thanks for what you do!

Most of the procedures in the MetarInfo module can be localized using .tcl files. However, the initial phase (icao_number "airport") word order is hard defined in MetarInfo.cpp.
`temp << "airports " << icao;
say(temp);
processEvent("say airport");
`

but in some languages (for example, in Russian) this word order is a mistake. Is it possible to transfer this part of the code to the logic of the .tcl, so that it was possible to set the pronunciation order in accordance with the rules of the language?

sm0svx, 2020-04-19:
Yes that seems wrong. The word ordering should not be hardcoded in C++. That is one of the main reasons TCL was chosen as an event handler so that word ordering in different languages can be supported.

@dl1hrc is the maintainer of the MetarInfo module.

dl1hrc, 2020-04-19:
ok, you may check my branch https://github.com/dl1hrc/svxlink/tree/MetarInfo-Announcement
The announcement is made in tcl procedure announce_airport and can be changed as own discretion.

73s de Adi / DL1HRC

sm0svx, 2020-04-19:
Ah nice. That was quick! Looks like a pull-request is coming :-)

circool, 2020-04-21:
Thank for all! Now everything is fine!
73!
Vladimir, R2ADU

closed 2020-04-21 by circool